hzeller/rpi-rgb-led-matrix #1460: Raspberry alternative, odroid maybe?

hakao32, 2022-09-02:
Hello
Recently, I have a problem about availability of raspberry pi in market . No stock..
I am looking for an alternative to raspberry. Is it possible to use odroid xu-4 to run this nice application? or do you have any alternative? Maybe a workarround to make it work with other platfroms.

anandrajgupta, 2022-09-02:
same problem with me,  maybe `Jetson Nano SBC`?
any suggestion will be appreciated, Thank you.

srcastor, 2022-09-19:
I think the best bet is to buy a CM4 and a [whaveshare Mini Base Board](https://www.waveshare.com/product/raspberry-pi/boards-kits/compute-module-4-cat/cm4-io-base-a.htm)

hzeller, 2022-09-19:
If you want to run tings on other platforms than Raspberry Pi, you need to read the datasheet of the SoC of your target platform and implement `gpio.h` and `gpio.cc`.
So the compute module suggestion is probably best: that is a standard Raspberry Pi, so it should work out of the box.

anandrajgupta, 2022-10-05:
Hi,
can you please give an idea if this program can be run on [Raspberry Pi Pico](https://www.raspberrypi.com/products/raspberry-pi-pico/)
This controller is very small and so is my application (like 3 or 4 LED Panels only).
And this controller's specs seems pretty much impressive according to the pricing:
Clock: 133 MHz
SRAM: 264kB 
GPIO Count: 30

Any suggestions?

If i get a positive hint/suggestion, i will start working on preparation of HAT with HC245 IC to shift the voltage level from 3.3V to 5V.

Please guide me in the right direction.

bluelasers, 2023-02-11:
The RP2040 is capable of supporting most LED Panels. However, there are some issues with it in standalone. RP2040 cannot do all processing in certain cases and probably does not have enough memory to support the approach used in this project. The approach used in this project is built around processors using DRAM. There is a means of using the Raspberry Pi and the Raspberry Pi PICO together.

This code base could support interfacing with the Raspberry Pi PICO. I do not know if it ever will or not. This would involve replacing the framebuffer.

As for the original topic, some boards have difference that have to be worked out with kernel logic. Overall, the approach used in this project would likely be the same.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828