hzeller/rpi-rgb-led-matrix #1461: Code not running in 64-bit Raspbian OS, any Solution?

anandrajgupta, 2022-09-09:
Hello Mr. @hzeller,

Thank you.

I was wondering if this code can be run on a Raspberry Pi 3 having a 64-bit Raspbian OS?
Directly running fails, any possible solution will be appreciated.

Thank you once again...

anandrajgupta, 2022-09-09:
As shown in the following picture, the compilation fails
[image]

anandrajgupta, 2022-09-09:
Also, if i try to run the pre-compiled code then it says "no such file or directory" (using sudo ./executableFile)

Note: I am trying to run it in 64-bit OS in a hope to get better frame-rate!

davemaster, 2022-09-09:
Greetings,
First, link or exact version of your 64-bit OS ?

hzeller, 2022-09-09:
It looks like you maybe copied over the version from a 32bit compile, then of course the object files don't work there.

First, `make clean`, then make again.

anandrajgupta, 2022-09-10:
Thank you very much, i will try...

anandrajgupta, 2022-10-05:
Thank you so much.
`make clean` worked!
Now it is able to run on 64-bit OS also, but no improvement in frame rate with it!
Thank you again.

closed 2022-10-05 by anandrajgupta