hzeller/rpi-rgb-led-matrix #1462: Split Image with HUB75E tried all options of hat/bonnets/electrodragon/custom one

ignaciogaticarojas, 2022-09-09:
Hi im having a matrix displaying image with a blan link in the middle because of hub75 E-Line.
Im curious why since i tried all ootions with the customized hat this repository suggest @hzeller  and also tried adafruit options and electrodragon. Doing the modifications and conections asked for each them. And i still have that blank line with allof them. Is there anyone in the community that could help with some advice? Thanks a lot [image]
[image]

hzeller, 2022-09-09:
looks like you have 40 pixel rows. Did you add `--led-rows=40` ?

ignaciogaticarojas, 2022-09-10:
That worked! @hzeller thank you again!

hzeller, 2022-09-10:
yay!

closed 2022-09-10 by hzeller