hzeller/rpi-rgb-led-matrix #1469: Help on reseting chip AHCT245 on active adapter

ignaciogaticarojas, 2022-10-01:
Hello dear friends of this amazing program.

I encounter a problem because i connected two panels from different batches and it seems that i desconfigured one of the AHCT245 chips from my active adapter. One hub75 output works normally but the first one does not.

Is there any way to reset a chip like this? or should i toss it out and replace with a new one?

Thanks a lot

anandrajgupta, 2022-10-01:
scenario unclear!

ignaciogaticarojas, 2022-10-01:
One hub75 of the active adapter doesn't shows images but the second one those. it seems there was a deconfiguration on that AHCT245 chip.

I had came across in the issues of the program with something called reset matrix.py

I wonder if this would do the trick,

anyone had have similar experience connecting two panels from different batches ? aka different chipset?

thanks

closed 2025-11-27 by marcmerlin