hzeller/rpi-rgb-led-matrix #1470: Can it run on Raspberry Pi Pico?

anandrajgupta, 2022-10-05:
Hi Mr. @hzeller,
Thank you for this wonderful program.

Can you please give an idea if this program can be run on [Raspberry Pi Pico](https://www.raspberrypi.com/products/raspberry-pi-pico/)
This controller is very small and so is my application (like 3 or 4 LED Panels only).

And this controller's specs seems pretty much impressive according to the pricing:
Clock: 133 MHz
SRAM: 264kB
GPIO Count: 30

Please give me some suggestions, meanwhile, I will start working on the preparation of a HAT (with 74HCT245 IC) for RPi-Pico to shift the voltage level from 3.3V to 5V.

Thank you to this wonderful community, feedbacks are welcome...

hzeller, 2022-10-05:
The Raspberry Pi Pico is an entirely different machine, this code won't work there. So using a Raspberry Pi Zero for about the same price is probably want you want.

If you want to use the Pico, maybe someone already build a program that you can use with that. I suspect the neat programmable GPIO ports of the Pico would make it very feasible.

anandrajgupta, 2022-10-05:
ok sir Thank you.

closed 2022-10-05 by anandrajgupta

nicholasgubbins, 2023-05-01:
hey @anandrajgupta did you get anywhere with a mapping library for the pico? looking at this now and keen to learn if you found a solution