hzeller/rpi-rgb-led-matrix #1471: Can we make taller display (>6 rows) without Rotating any Panel upside down?

anandrajgupta, 2022-10-11:
Hi Mr. @hzeller,
First of all, Thank you.

There is a problem if we rotate an LED panel upside down, water enters the display!

This is because there are few holes in the bottom side of individual LED matrix Panel which is designed for water outlet in case some minor amount of water enters the Panel and there are no such holes in the top side of Panels to prevent rainwater from entering the Panel from top,
Now, if we rotate the Panel (to use "U-mapper" for taller display having row count greater than 6) then the bottom side comes up thus the bottom holes also come up which causes a lot of water to enter in the Display and no holes in bottom to let the water out from the Display!
That's why I am thinking of a solution to accomplish taller display without turning any Panel upside down, is it possible anyhow? 
Note: "V-mapper" can't be used because panels per row is > 10

I am using Compute-Module 3
I have prepared Active-6 HAT
Everything is working fine, it's just the water inlet due to inversion of LED Matrix Panels
What I am trying to achieve can be clarified by the following Diagram:
[image]

Please let me know if this is possible anyhow.

Note: I have an idea to accomplish this, but I don't know how to implement that, I am thinking to connect an FRC from the "Output connector of last Panel of 1st Row" to the "Input connector of 1st Panel of 2nd Row" shown in below Diagram, please have a look and help me if possible:
[image]

Thank you very much to this awesome community...

davemaster, 2022-10-11:
Did You tried? Pictures, 
- define screen width, height
- parallel = 3
- chain = 6

try this, then picture 

NOTE: do You tried your panels individually? 

Best regards

anandrajgupta, 2022-10-11:
thank you for your time.

are you sure you read my problem carefully...

davemaster, 2022-10-11:
> 

Are You sure you read the entire options of this github carefully?

anandrajgupta, 2022-10-11:
please don't get offended, just read this issue title and you will get to know my problem, thanks.

davemaster, 2022-10-11:
3 parallel, not possible (I'm saying that above) BUT, You can programme it, so you're software reorder the image, or build your own mapper

https://sites.google.com/teletrackingsac.net/teletrackingsac

anandrajgupta, 2022-10-11:
my project need to have 12 rows and 15 columns (a total of 12x15 = 180 panels
and i need an arrangement in which none of my panels are upside down...

davemaster, 2022-10-11:
resolution of your panels?

anandrajgupta, 2022-10-11:
16 × 32

anandrajgupta, 2022-11-05:
@hzeller

ledvinap, 2022-11-05:
@anandrajgupta My PR should support it (kind of): https://github.com/hzeller/rpi-rgb-led-matrix/pull/1478

But it is only slightly tested, ping me if you need help.

You will probably have problem with long FRC cables, so some zig-zag arrangement may be necessary

anandrajgupta, 2022-11-05:
Thank you very much, i will give it a try ✌🏻

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828