hzeller/rpi-rgb-led-matrix #1476: Help understand pixel timings how 100Hz is possible

Reider, 2022-11-01:
I'm trying to understand exactly how the 11 bit color is possible with these displays but my math is off somewhere. 

For example if we have this panel: 
https://www.aliexpress.com/item/32518185271.html
It is a 64*32 pixel panel that has DP5020B drivers that has a max speed of 25MHZ
https://datasheetspdf.com/pdf/1391447/DP/DP5020B/1

The constant current drivers are chained in a row per RGB on the top and RGB on the bottom. R1,R2,G1,G2,B1,B2 Each row is 64 pixels so we have to clock in 64 bits of data at 25Mhz for 2 lines simultaneously, top and bottom when selecting the line.
We have to do this 16 times per 2 lines so 32 rows. That's 2048 pixels, 40ns per pixel.
If we want 11 bit plates and each plate time has to double that's 2048 ticks for 11 bitplates.

40ns * 2048(pixel) * 2048(plates) = **167.77 milliseconds per frame**. That would be aprox **6** frames per second.

The documentation states 100Hz refresh rate so I know I'm off somewhere. 
Please help

ledvinap, 2022-11-01:
Intensity code is sent as binary - shift in column data for given bit (RGB*2 in parallel), than generate row pulse with length given by bit position (2^bit time units).
So it's 'only' 40ns * 2048(pixel) * 11(bits/color)

Reider, 2022-11-01:
So the 11 bit plates are on a different delay clock and take up 11 * 40ns time that is 440 ns.
For the Most significant bit on the plate 220ns after 110ns after 55ns and so on... Because the MSB needs to show up more to indicate intensity. For the smallest plates that seems to be sub ns delays. Is the RPI capable of that, or am I misunderstanding something?

ledvinap, 2022-11-01:
Frame display algorithm:
<pre>
for d_row in [0 .. double_rows):
  for b in [0 .. kBitPlanes):   # bit loop
   shift `columns`  (RGB * 2 rows in parallel) column bits into display, sending data for doublerow `d_row`, bitplane `b`
   select doublerow `d_row` (changed only after iterating all bitplanes, using ABCD panel inputs)
   generate strobe pulse with length = 2^b * pwm_lsb_nanoseconds
</pre>

in your case:
double_rows = 16
columns = 64
kBitPlanes = 11
pwm_lsb_nanoseconds = 130 default, parameter `--led-pwm_lsb_nanoseconds`

Led on time per frame is (2^12-1) * 130ns ~ 530us per double row, * 16 double_rows = 8.5ms, 117Hz
Shift time is 40ns * columns * kBitPlanes * double_rows = 40ns * 64 * 11 * 16 = 450 microseconds, but shift is actually done when previous (latched) double_row is displayed, so it's influence is actually smaller. Worst case it will decrease update rate to ~110Hz
Some time is spent in other necessary delays, but 110Hz seems attainable

PWM timer is normally used for strobe pulse (but correct pinout is necessary for this). There is fallback available that uses `nanosleep` + busy wait

Reider, 2022-11-01:
Ah everything is clear now. Thank you for your time

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828