hzeller/rpi-rgb-led-matrix #1481: Adafruit hat broke current LED Matrix Setup

banning4, 2022-11-20:
@hzeller 
I have a single chain of seventeen 16x32 led panels. I had previously wired my raspberry pi 3b+ to the first panel using jumper cables per the wording diagram, and it worked great. I have the panels hooked up to 3 separate 5V 15a power supplies with 6 displays on each power supply (I use to have 18 but one broke). So my current issue isn't related to power, as that setup worked perfectly. Yesterday, I replaced my wiring with an adafruit hat for my pi. For some reason, now only the first 6 panels in my chain are turning on (while using the new gpio flag) and they look very dim and flickery. These 6 panels work whether or not their power supply is plugged in so they are being powered by the power supply from the board? I have not hooked up the 5v output from the board to anything. Is this a problem with the GPIO mapping on the hat? is the data being sent out to the chain somehow different than before?

anandrajgupta, 2022-11-20:
What exactly is the issue 🤔

ledvinap, 2022-11-21:
@banning4 : It seems that panels are powered only by rectified data signals from hat. Check 5V power supply to panels.

davemaster, 2022-11-21:
Also check each panel, one by one again.