hzeller/rpi-rgb-led-matrix #1483: Can't get rid of erratic white lines on side of screen and trails of color

popestr, 2022-12-13:
Hello, and a preemptive thank you to whoever tries to help me get to the bottom of this!

First off, here is a video of the behavior: https://drive.google.com/file/d/1nrma7VP-nIduiQNT1083UQ4YT7h31eHd/view?usp=sharing

I have manually wired the connections using female-female breadboard wires from the panel to a Raspberry Pi Model 3B and am using a 5V/8A power supply. Unfortunately I have not been able to get Demo 0 or any other demo working properly no matter what I try. Every demo shows these white / cyan lines on the left side of the display, as well as streaks throughout the entire display.

I have confirmed that the wiring is correct, and have tried various levels of current, both a Raspberry Pi Zero and a 3B (both running DietPi) and I can confirm that the display works properly with another microcontroller running Micropython (Adafruit Matrix Portal M4).

The problem seemed to get worse with lower currents, which is why I went with a 5V/8A power supply thinking that high current would resolve the issue (it didn't). I have tried all variations of `led-multiplexing`, `led-row-addr-type`, `led-slowdown-gpio` and `led-no-hardware-pulse`, all with no effect on the noisy output. I purchased the panel from Adafruit, and so I would expect that this panel has been used in this project successfully by now -- unfortunately I cannot find any similar issues.

The video attached shows the issue with Demo 0, but this issue exists for all demos available.

anandrajgupta, 2022-12-13:
There is a HAT requirement between Raspberry Pi and Panel's HUB75 connector.
If you are using loose jumper wires, then you would need 74HCT245 Level Shifter IC which shifts 3.3V logic of Raspberry Pi GPIO to 5V logic of RGB Panel.

popestr, 2022-12-13:
Thank you so much for the response! I had read somewhere that there was potential for the 3.3V control signal to work, but it seems for my panel this is not the case. I will close this issue while I wait for some new parts to arrive.

Cheers!

closed 2022-12-13 by popestr

hzeller, 2022-12-13:
Also
 * Make sure you have ground connected between the panel power supply and the signal GND of the Pi.
 * Your panel might be too slow compared to your Pi. Test different values for `--led-slowdown-gpio` to see if it fixes it.

anandrajgupta, 2022-12-13:
yes, definitely try
--led-slowdown-gpio = 5

btw @popestr you already mentioned that you tried this option, i hope so...

popestr, 2022-12-13:
Henner, thank you! You have just saved me $30 and a headache! Simply using an extra ground pin on the Raspberry Pi and connecting it to the power supply's ground output has completely resolved the issue. Everything looks perfect now!

My sincerest gratitude for your thoughtful and timely reply. And another thank you for the open-source software! Have a fantastic holiday season my friend.

Ryan