hzeller/rpi-rgb-led-matrix #1485: Word Wrap Auto

Holychris56, 2022-12-15:
Hello! Is there a way to do auto word wrap on an LED screen, instead of trying to find out what is the best font/size/offsets?

I am using the python library.

Thanks!

anandrajgupta, 2022-12-15:
the drawText function returns how much pixels are used in writing the word

Holychris56, 2022-12-15:
How does that allow me to have an auto wrap? I'm running a 64x32 led matrix. I would like to write "hello world" and when it get cuts off, it writes the following world to the next line.

anandrajgupta, 2022-12-15:
you will have to make your logic to achieve that, using the feature mentioned above

Holychris56, 2022-12-17:
ahhh ok. I thought there was already code written to do this. Didn't want to re-invent the wheel haha!

closed 2022-12-31 by Holychris56