hzeller/rpi-rgb-led-matrix #1489: is it possible to use with a rockpro64 from pine64 with a rk3399

hannescam, 2022-12-23:
i want to use this amazing library with the rockpro64 becaue all of my raspberries died

anandrajgupta, 2022-12-24:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1460

hannescam, 2022-12-24:
so is it possible to run this code with a led panel on a rk3399?

hannescam, 2023-01-30:
@anandrajgupta i tried and there was a not a raspberry pi error and a premission error

ledvinap, 2023-01-31:
GPIO implementation is quite different on rk3399, there is no SET/CLEAR register and 3 IO blocks are connected to 40pin header. High-performace implementation will be more difficult and cooperation with other software using GPIO access will be problematic.

hannescam, 2023-01-31:
and what about the allwinner a64 or d1

ledvinap, 2023-01-31:
Both have very similar GPIO implementations as Rockchip ...

hannescam, 2023-01-31:
what other options are ecxept than a raspberry pi that could work with this software with little/no modification

ledvinap, 2023-02-01:
Seems that Broadcom has quite unique GPIO implementation, probably caused by CPU being far prom GPIO (GPU is first-class citizen).
rk3399 should be possible, at least as proof-of-concept version. It should be quite easy to rewrite current code to unpack framebuffer data and write it into corresponding registers. Just check usage of other IO in used banks (only GPIO OUT mode is important)

hannescam, 2023-02-01:
cool it would be great if this project could be opend up to other platforms (not just these stupid expensive fast braking things i broke 5 allready) but i am not smart egnuth to implement this

bluelasers, 2023-02-11:
There may be a way of working around the GPIO issues with a kernel module. However, the GPIO does work differently. This code base uses a kernel module to gain access to the GPIO or bypasses with /dev/mem. 

Has anyone tried using /dev/mem with GPIO to see what the performance is on RockChip?

ledvinap, 2023-02-11:
@bluelasers : this project does access GPIO directly, both in /dev/gpiomem and /dev/mem case.
Problem is cooperating with kernel (or other) code that uses pins in GPIO mode. But is quite possible that no such pins are vital for this use case (pins that use other GPIO mux setting are fine).

hannescam, 2023-02-11:
i think with the kernel module it will be too slow 
i didnt ever develop any kernel modules so correct me if i am wrong

andytheengineerguy, 2023-07-19:
Hi @hannescam, any update on this 5 months later? I am running into the same issue using a pine64 OSQuartz. Would love to use a raspberry pi compute module but still, none are available :(

andytheengineerguy, 2023-07-19:
@ledvinap, did you ever try the proof of concept version for any rockchip CPU?

ledvinap, 2023-07-19:
@andytheengineerguy : Sorry, I didn't even try, I have no use for RK boards.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828