hzeller/rpi-rgb-led-matrix #1490: how to display two lines at the same time

suibari, 2022-12-26:
I use a 64x32 LED matrix panel and Raspberry Pi Zero WH.
I could run examples with no flickering.
So I am trying to show two lines of text.
As you posted in issue #404, I have attempted to display two lines of text with the following command.

```
$ sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/scrolling-text-example -s0 --led-cols=64 --led-rows=32 --led-slowdown-gpio=0 -f /home/pi/Downloads/font/misaki_bdf_2021-05-05/misaki_gothic_2nd.bdf "hello world" &
$ sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/scrolling-text-example -s0 --led-cols=64 --led-rows=32 --led-slowdown-gpio=0 -f /home/pi/Downloads/font/misaki_bdf_2021-05-05/misaki_gothic_2nd.bdf "hello world" -y9 &
```
(The font I'm using is an 8x8 font)

As a result, it does not work as intended, and occurs character misalignment and screen flickering.
How can I display two lines?

hzeller, 2022-12-26:
you can;t run two programs at the same time, they fight for the same hardware.

So in order to achieve your goal, you've to modify the sample program to show two lines.

suibari, 2022-12-27:
Thank you for your prompt reply.
I tried what you said and it just worked!
Thank you so much.

[image]

closed 2022-12-27 by suibari