hzeller/rpi-rgb-led-matrix #1494: Multi Thread Safe?

Holychris56, 2022-12-31:
Hello! 

I am trying to write to the LED screen with multiple threads. I do get some flickering when multiple threads are writing to the led matrix. Am I doing something wrong or has this not been tested before?

For example, I have a clock that gets updated on one thread, while weather data, which is scrolling, gets updated on a different one. They both use the led matrix API to write the data.

Thanks

anandrajgupta, 2023-01-01:
c or python?

Holychris56, 2023-01-01:
Python

Holychris56, 2023-01-01:
Python

hzeller, 2023-01-01:
I would _not_ recommend using multiple threads to update the currently active canvas. Setting a pixel requires multiple writes internally and it can not be protected by a mutex, as it would interrupt the thread constantly refreshing the display.

Instead, make the granularity of updates for a full frame. This is commonly used as well in regular graphics to avoid tearing.

Create an off-screen canvas, fill it with the next frame and then `SwapOnVSync()` that in. That way, you will not have any glitches or tearing. Also, you don't need multiple threads to update the content, you can prepare the entire next frame of your animation in the main thread (draw the clock, draw the weather data, write the text to the next scroll position). Simplifies the code a lot.

You also will have exact timing as the `SwapOnVSync()` changes the off-screen canvas atomically when switching to the next frame.

This is true for C and Python (in particular Python, as that is way slower and thus emphasizes glitches)

Holychris56, 2023-01-01:
I was using the PIL library for this. Example of my code:

`    matrix = rgbmatrix.RGBMatrix(options=options)
      canvas = Image.new('RGB', (matrix.width, matrix.height))
      draw = ImageDraw.Draw(offscreen_canvas)
      matrix.SetImage(offscreen_canvas)
`

I was then writing texts like this:

` draw.text((0, 12), "time", utils.Color.TEAL, font)`

However, IF I add an if statement, in a while loop to have a scrolling effort, the draw.text does NOT get written to the LED screen until the entire scrolling is completed off-screen, which will then write the draw.text (which was in the if statement). 

I pretty much have a conditional statement in a while loop, once that statement is true, I want to write to the LED screen. However, it seems the API won't allow the data to be written until there are no more LED lights being powered. After that, it writes to the LED screen, where the conditional statement is.

Holychris56, 2023-01-02:
I removed all of the PIL code, and changed it to the graphics.DrawText API. The scrolling now works with different colors.

closed 2023-01-02 by Holychris56

anandrajgupta, 2023-01-02:
can you show the sample now 😃

Holychris56, 2023-01-02:
` graphics.DrawText(self.canvas, self.textFont, secondPostion, 18, utils.GraphicsColor.TEAL, data['text']) `
` graphics.DrawText(self.canvas, self.textFont, secondPostion, 25,  utils.GraphicsColor.GREEN if 
   data['value'] > 0 else utils.GraphicsColor.RED,   str(data['value'])) `

     ` secondPostion = secondPostion - 1 `

                      
  `  graphics.DrawText(self.canvas, self.textFont, xPosition, 18, utils.GraphicsColor.TEAL, data2['text']) `
    ` graphics.DrawText(self.canvas, self.textFont, xPosition, 25, 
                                utils.GraphicsColor.GREEN if data2['value'] > 0 else utils.GraphicsColor.RED,  
                                str(data2['value'])) `

   ` xPostion = xPostion + 1 `
   ` self.canvas = self.matrix.SwapOnVSync(self.canvas) `

ledvinap, 2023-01-02:
@Holychris56
It should be possible to use PIL. Something like:
``` python
   matrix = rgbmatrix.RGBMatrix(options=options)
   offscreen_canvas = matrix.CreateFrameCanvas()

   image= Image.new('RGB', (matrix.width, matrix.height))
   draw = ImageDraw.Draw(image)
  
   while True:
      # use draw to generate image
     draw.text((0, 12), "time", utils.Color.TEAL, font)
     ...
     # copy image to offscreen canvas
     offscreen_canvas.SetImage(image)
     # swap canvas to display it
     offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)

anandrajgupta, 2023-01-02:
i meant Display graphics sample 😅