hzeller/rpi-rgb-led-matrix #1496: Text and some pictures are one pixelish off

pbullion, 2023-01-05:
The bigger the font and full size pics are coming out like in this video where the top half is a pixel or two off and is kind of split.

This happen to anyone else? Or happens to know the root problem and or a fix?

Thanks!

https://user-images.githubusercontent.com/20978249/210891739-78c907c2-28a2-4bd5-9a0b-27f09fd3b471.MOV

hzeller, 2023-01-05:
This looks a little bit like tearing.
How do you output the text ? Are you using the `text-scroller` utility or wrote it yourself ? If the latter, are you using the frame-canvas and `SwapOnVSync()` ?

pbullion, 2023-01-05:
Its using python bindings and yes the createframecanvas and the swaponvsync

hzeller, 2023-01-05:
Do you see the same thing when the text is not scrolling ?
Are you using bdf fonts with the `DrawText()` function or try to render it from some ttf with some other font renderer (which usually works badly for low resolution).

anandrajgupta, 2023-01-06:
FRC length might be causing this delay in top half and bottom half

hzeller, 2023-01-06:
Something unrelated to the font issue: make sure to not create frame canvases in the loop. The `offscreen_canvas = CreateFrameCanvas()` line should be before the `while True:`.
Otherwise you run out of memory soon.

hzeller, 2023-01-06:
Also, the whole font-loading is probably also not something you want to do in every loop, but move it to the front,

pbullion, 2023-01-06:
So i went back and just ran the straight runtext.py in the samples folder. still getting the same thing

[image]

https://user-images.githubusercontent.com/20978249/211073605-59b15d73-e0ac-4aa5-9b9b-86f7ca1bde95.MOV

https://user-images.githubusercontent.com/20978249/211074227-90a61b46-839a-49fd-b7a4-c0786ecbac13.mov

pbullion, 2023-09-11:
@hzeller any idea on this? just running the runtext.py example straight from you is still showing the pixels off in the middle

utemorley2012, 2024-11-02:
Was there ever a resolution here? I am seeing the same issue. The tearing gets worse the faster I set the image to scroll. I am using the python image-scroller sample program.