hzeller/rpi-rgb-led-matrix #1497: Clear half the display

ChopsKingsland, 2023-01-07:
I am currently clearing the whole display just to update the right hand side, which is resulting in the attached video on the left hand side of the display... is it possible to only clear the right hand side?

https://user-images.githubusercontent.com/18499854/211169479-3967caa5-af57-4423-adf4-6c28ecb62227.MOV

ChopsKingsland, 2023-01-07:
I worked it out... just drew black text over the top of the text that I wanted to get rid of. However, I still get glitches with the image... video attached (the horizontal lines, diagonal are from camera frame rate)

https://user-images.githubusercontent.com/18499854/211170696-1085629b-8f53-4228-912e-979907f46f2a.MOV

ledvinap, 2023-01-10:
Simply use double-buffering (`SwapOnVSync` )