hzeller/rpi-rgb-led-matrix #1498: Crash during rgb_matrix::RGBMatrix::CreateFromOptions(...)

hexabyte23, 2023-01-07:
Hi all,

I know it is a very basic scope pointer problem, but I am stuck.
Shame on me, but I have no more idea... since months !

I try to replace this code :

```
RGBMatrix::Options global_options;

int main(int argc, char **argv) {
   global_options.hardware_mapping = "regular";
   // ...
}
```
by a bit more flexible one :

```
RGBMatrix::Options global_options;

int main(int argc, char **argv) {
   std::string myVar = "regular";
   global_options.hardware_mapping = myVar.c_str();
   // ...
}
```
but app crash during call of rgb_matrix::RGBMatrix::CreateFromOptions(...)
even this version crash !

```
RGBMatrix::Options global_options;

int main(int argc, char **argv) {
   std::string myVar = "regular";
   global_options.hardware_mapping = strdup(myVar.c_str());
   // ...
}
```

Any thoughts ?

hzeller, 2023-01-07:
Why do you need the options globally ?
Where do you call CreateFromOptions() ? Is it directly after you set the global options, or do you have some sort of situation where the lifetime is more muddy ?

hexabyte23, 2023-01-08:
Moving options globally is just make sure that pointer will be valid anywhere.
No, I dont call CreateFromOptions() directly after, what I do not understand is even I allocate memory by using strdup(), my program crash.

I tried to replace 

```
class RGBMatrix : public Canvas {

const char *hardware_mapping
...
}
```

by

```
class RGBMatrix : public Canvas {

std::string hardware_mapping
...
}
```

But that imply too many works...

hzeller, 2023-01-08:
If you put the options globally, you have all kinds of other lifetime situations to worry about. Why can't create the options and use right away like in all the examples ?

hexabyte23, 2023-01-08:
Good question, I never read examples, I will do.

hexabyte23, 2023-01-14:
I finally succeed to have options.hardware_mapping pointing to « regular » or « adafruit…" string by doing this:

'''
const char* regular  = "regular";
const char* adafruit = "adafruit-hat-pwm";
rgb_matrix::RGBMatrix::Options options;

int main(int argc, char **argv) {
  std::string myVar = "regular";  // from a config file
   
  if(myVar == "regular")
     options.hardware_mapping      = adafruit;
  else
     options.hardware_mapping      = regular;

   // ...
}
'''

a bit sad, but it's working that way...

hzeller, 2023-01-14:
If you can't use `myVar.c_str()` directly, then it sounds like your `myVar` goes out of scope before `CreateFromOptions()`.

hzeller, 2023-01-14:
(also, you still have your options as global variable, which indicates you're attempting to do something that is prone to lifetime issues)