hzeller/rpi-rgb-led-matrix #1499: P4 80x40 1:10 display integration issue

priya2212, 2023-01-09:
Hi All, 

I have followed [the article](https://github.com/hzeller/rpi-rgb-led-matrix/blob/c2c7280cf2b3046457e79332cd22b0ac70b3621d/wiring.md) to wire up the P4 LED with RPI. I am facing issues in writing text and scrolling images. 
I have run the code using below command - 
`sudo examples-api-use/text-example -f fonts/helvR12.bdf --led-cols=80 --led-rows=40 --led-multiplexing=0  --led-no-hardware-pulse`
changed it to` sudo examples-api-use/text-example -f fonts/helvR12.bdf --led-cols=16 --led-rows=40 --led-chain=5 --led-multiplexing=0  --led-no-hardware-pulse` 
But unfortunately getting the same results attached in the image. 

I have gone through these forum links - [Different Scan Rates](https://github.com/hzeller/rpi-rgb-led-matrix/issues/49) /[ P4 80x40 10S (1:10 scan) multiplexer help](https://github.com/hzeller/rpi-rgb-led-matrix/issues/986) and tried adding own multiplexer as recommended but not able to run the example. 

Any help will be appreciated. If you need any information from my end, I would be happy answering them as early as possible. 
[image]
[image]

priya2212, 2023-01-11:
Anybody can please help me. 
I got the module specification from manufacturer - 
[image]

priya2212, 2023-01-17:
Hi @anandrajgupta, Thank you very much for your response. I am currently away from the system and would give it a try very first thing. Just to mention, I have connected E as well and follow the wiring from here -  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

b4rtek, 2023-01-17:
Hey @priya2212 
you must modify `lib/multiplex-mappers.cc` by adding new mapper 
```
class P4Outdoor80x40 : public MultiplexMapperBase {
public:
  P4Outdoor80x40() : MultiplexMapperBase("P4Outdoor80x40", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int shift = (y  / (panel_rows_ / 4)) % 2;
    *matrix_y = ((y / (panel_rows_/2)) * (panel_rows_/4)  + y % (panel_rows_/4));
    *matrix_x = ((x * 2) + (shift ? 0 : 1));
  }
};
```
(like in (https://github.com/hzeller/rpi-rgb-led-matrix/issues/49) and recompile. When u do this u are be able to use 
`--led-multiplexing=19` Thats should help. I see u have almost same board as my, only yours is v2.0 and i have v1.0. I this this is irrelevant but i am not an expert ;)

priya2212, 2023-01-18:
Hi @anandrajgupta and @b4rtek , 

I tested the library with your mentioned suggestions and following are the results -

Test 1. Tested it with - led-multiplexing=1 and ran the following command - 
```
sudo ./examples-api-use/text-example --led-brightness=100 --led-cols=80 --led-rows=40 --led-chain=1 --led-pwm-lsb-nanoseconds=300 --led-slowdown-gpio=5 -f fonts/helvR12.bdf -C 255,0,0 --led-multiplexing=1
Enter lines. Full screen or empty line clears screen.
Supports UTF-8. CTRL-D for exit.
HELLO
HELLO
```
[image]

The strings are displayed in the first half only, not second half and due to this, when writing 2nd hello, the string got cut. 

Test 2. Tested with - led-multiplexer=19 and ran the following command - 
```
sudo ./examples-api-use/text-example --led-brightness=100 --led-cols=80 --led-rows=40 --led-chain=1 --led-pwm-lsb-nanoseconds=300 --led-slowdown-gpio=5 -f fonts/helvR12.bdf -C 255,0,0 --led-multiplexing=19
Enter lines. Full screen or empty line clears screen.
Supports UTF-8. CTRL-D for exit.
HELLO
```
[image]
Attached photo for git diff for your reference.
[image]

However, with led-multiplexing as 1, I got some positive results as with the python binding, I was able to scroll the text in the middle - 
`sudo python runtext.py  --led-brightness=100 --led-cols=80 --led-rows=40 --led-chain=1 --led-pwm-lsb-nanoseconds=300 --led-slowdown-gpio=5 --led-multiplexing=1`

https://user-images.githubusercontent.com/20925260/213113474-592f6d27-355b-427c-83cb-b4a37a059c42.mp4

Can we change something in led-multiplexing=1 to make it work as required?

closed 2023-10-29 by priya2212