hzeller/rpi-rgb-led-matrix #1500: 128*64 outdoor panel

OkSingra, 2023-01-10:
Help connect the panel, does not respond to the connection. I've already tried all the options. Direct connection and through bonnet HAT do not give a result.
[image]
[image]

wata-net, 2023-01-10:
What are the part numbers for the row driver chip and column driver chip?

davemaster, 2023-01-10:
Greetings,

You have the HUB75 connections in the picture here.... check the connections again, THEN, check the power supply, this panel has a lot of RGB LEDS, try with a 20A or greater power supply, and only ONE PANEL... then try ALL THE options --led-multiplexing: --led-multiplexing=<0..17> : Mux type: 0=direct; 1=Stripe; 2=Checkered...

best regards

ledvinap, 2023-01-10:
The DI marked pin on HUB75 is very suspicious. I doubt this panel will work without feeding some data into it.
Try reading part numbers on chips. 
And take look at https://github.com/hzeller/rpi-rgb-led-matrix/blob/dfc27c15c224a92496034a39512a274744879e86/lib/framebuffer.cc#L121, this one at least mentions DIN pin.

IMO, at minimum, you need to connect DI pin to E signal (jumper on adapter)

OkSingra, 2023-01-11:
[image]
I tried various options, and could not achieve the glow of the LEDs.

anandrajgupta, 2023-01-12:
--led-panel-type=<name>   : Needed to initialize special panels. Supported: 'FM6126A', 'FM6127'

Tadont, 2023-01-26:
@OkSingra did you solve the problem with this panel?

OkSingra, 2023-01-30:
[image]
[image]
So far I have only been able to achieve this.
sudo ./text-scroller -f ../fonts/9x18.bdf --led-gpio-mapping=adafruit-hat --led-rows=64 --led-cols=128 --led-slowdown-gpio=5 "Hello World ♥"

Tadont, 2023-01-31:
@OkSingra I have a similar panel and a similar hat. I can't display anything