hzeller/rpi-rgb-led-matrix #1502: Enhance API

hexabyte23, 2023-01-14:
Hi all,

My understanding, whatever 'high level ' DrawText() or SetImage() functions we use, all is resumed by massive calls to canvas->SetPixel(), right ?

If so, as many 'high level' functions are missing (ellipse, rounded rectangle...) is it a way to directly give final RGB array (provided by another canvas framework) to rgb_matrix::Framebuffer class, without using too consuming loop in loop :

```
...
   for (int y = canvas_offset_y; y < h; ++y) {
      for (int x = canvas_offset_x; x < w; ++x) {
        c->SetPixel(x, y, buffer[0], buffer[1], buffer[2]);
        buffer += 3;
      }
      buffer += next_row_skip;
    }
...
```

hzeller, 2023-01-14:
Currently there is no API like that. Reason is two-fold
  1. The `Canvas` interface should be as simple as possible so that it is fairly easy to generate a new wrapping implementation. 
  2. There would not be much performance advantage: setting a pixel is fairly expensive as it prepares the memory to be dumped to the matrix, so it is decomposing the value to bitplanes with multiple memory writes. If there was a block-setting function, it could not simply just `memcpy()` and be done with it, it would need to set each pixel individually as well. Bottom line: having a loop outside the Canvas is not really a problem. Setting pixels can only be done with a few megapixel/sec speed on the Pi.

If you want to pre-generate frames and swap out easy for animations etc, I suggest to have a look at the `content-streamer.h`.

hexabyte23, 2023-01-15:
very clear, thanks Henner