hzeller/rpi-rgb-led-matrix #1504: LED display 128x64 RPi4 RGB Matrix Hat problem

Tadont, 2023-01-26:
Hello! I connect this LED screen using raspberry pi4 and  RGB Matrix Hat. using config data. None of the examples run(Python, demo, led-image-viewer). would be grateful for help
[image]
[image]
[image]
[image]

https://user-images.githubusercontent.com/123635181/214839035-d818547f-5356-4c44-a274-339f1b7ec16e.mp4

`sudo ./led-image-viewer --led-cols=128 --led-rows=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=1`

anandrajgupta, 2023-01-26:
1st
--led-slowdown-gpio=5

2nd
led image viewer will take image as an argument

3rd
led multiplexing will have to be applied

etc...

put --help in the command line argument and you will get all possible options

Tadont, 2023-01-26:
--led-slowdown-gpio=5 produced such a result. led multiplexing doesn't change anything. Maybe some more advice?
[image]

anandrajgupta, 2023-01-27:
i think it is `--led-gpio-mapping` issue, you might need to choose something else because the content is not being drawn at all!

led multiplexing command is `--led-multiplexing=x`
where x = 0 to 17 digit

Tadont, 2023-01-27:
@anandrajgupta, Unfortunately, none of the above settings help.

Tadont, 2023-01-27:
the chip that my rgb matrix uses: SM16169S

anandrajgupta, 2023-01-27:
same chip https://github.com/hzeller/rpi-rgb-led-matrix/issues/1015

wata-net, 2023-01-27:
Thank you for letting me know the model number of the driver IC.
According to the datasheet,
"SM16169SD is a high-brush constant-current drive chip using SM-PWM protocol. It has a built-in 8K SRAM memory, which can store a complete frame of display data, and supports 2~16 scan LED display screens to store and display the entire frame of data."
So this library probably does not support SM16169S.
I don't know if it's correct, but I tried to find out how to initialize the SM16169. (I'm assuming RT5958 because I'm not sure about the row driver.)
I hope someone sees this and responds...
[image]

Tadont, 2023-01-27:
@wata-net DP7269 number of the second chip on the panel

Tadont, 2023-01-27:
@wata-net DP7268 number of the second chip on the panel

wata-net, 2023-01-27:
Judging from the model number DP7268, it is probably a product of "德普微电子".
But I couldn't find out the details.
The model number is similar to TC7258, so it may be the same product.
By the way, TC7258 is a normal 3 to 8-Line Decoder, so no special settings should be required.

Tadont, 2023-01-27:
But for some reason it doesn't work

wata-net, 2023-01-27:
The DP7268 does not require any special settings, but the SM16169 requires a special initialization procedure.
If the DP7268 works fine and the SM16169 does not, the LED panel will light up but the correct image will not be displayed.
Both DP7268 and SM16169 need to work properly to display correct image.

andytheengineerguy, 2024-01-04:
@Tadont or @wata-net, did you ever find out how to initialize the [SM16169SH](https://www.linkage.cn/uploads/allimg/20231204/2-231204161635920.pdf)?

I am facing the same problem now. I have translated the datasheet into English, and in general, I understand it. But I'm yet to figure out how this library does the 'initialization sequence' of the panel.

bsparacino, 2024-01-10:
I keep running into this issue too, manufacturers keep updating their modules with SM16169SH chips and I can't use them. I have 4 different modules right now with this chip that I can't use. I will send a monetary reward to whoever comes up with a solution.

andytheengineerguy, 2024-01-10:
@bsparacino I feel you!!! I have a **lot** invested in this library and the project I am basing it on. I am very willing to compensate whoever solves this too. @hzeller is this cheeky to ask? Could you be this person..?! 😅

bsparacino, 2024-01-10:
And now I just got an order of 400+ modules in with ICND2153AP chips, of course the manufacturer changed the design from our previous order and didn't tell us. Now I have a pile of rocks I can't use.

wata-net, 2024-01-10:
I don't have an SM16169 panel, so I can't confirm the operation.
Has anyone tried the previously posted analysis results?

andytheengineerguy, 2024-01-10:
@wata-net thank you for posting the analyses results - personally, this is beyond my skill level in terms of firmware development. Do you know anyone who could translate your analyses into a C initialization sequence to integrate into this library?

wata-net, 2024-01-12:
@andytheengineerguy Sorry, I don't know.
If I'm lucky enough to get an SM16169 panel, I'll report back.
Since only STM32 can be used, I plan to check the operation with STM32.

bsparacino, 2024-01-12:
@wata-net I have SM16169 panels, I can mail you a couple to test. I'm in the US.

andytheengineerguy, 2024-01-12:
@bsparacino and @wata-net I also have spare SM16169 panels that I'm willing to ship. Let me know if I can be of any use here 💪

wata-net, 2024-01-13:
I have uploaded the HUB75 driver project file for NUCLEO-F446RE.
[https://drive.google.com/file/d/10cotodkM1iLoY0CljS72_RwHh7dI84Kq/view?usp=sharing](url)
Please load it into STM32CubeIDE and use it.
Compatible driver chips are SM16380, ICND2153, and SM16169.
SM16380 and ICND2153 have been confirmed to work.
Could someone please confirm the operation of SM16169?

Please specify the number of pixels of the LED panel in lines 230 to 236 of main.c.
Also, specify the type of driver chip on line 309.

Everything is written in main.c. Please decipher.
However, my coding skills are crap. please forgive me.

If you do not want to convert the level of the output of NUCLEO-F446RE, please set the power supply voltage of the LED panel to 4V.
This is very important.

I am in Japan.
Thank you for suggesting sending LED panels.
However, it costs money, so I would like someone to try the above project file.

[image]
[image]

andytheengineerguy, 2024-01-16:
Hi @wata-net, thank you for sharing this. I have ~intermediate experience with both the Nucleo and STM32CubeIDE. I have used them both at work, but do not own any Nucleos at home. However, with a bit of time and some guidance from you, I'm confident I could test this. 

Some questions:
1. Is there any way you could adapt this code to a fork of the rpi-rgb-led-matrix so that I can test it directly on RPi hardware? If not, no worries, I would be willing to buy a Nucleo to test your code. Let me know.
2. Just curious - you mention the compatible driver chips as SM16380, ICND2153, and SM16169, SM16380 and ICND2153. Is it because they all share the same initialization sequence?

wata-net, 2024-01-16:
@andytheengineerguy
1. I've never used a raspberry pi.
I hope someone does it.

2. These three driver chips are electrically the same, but there are some differences.
For example, the control commands to send, the number and contents of configuration registers, the number of GCLK pulses, etc.
These three were successfully controlled by analyzing signals from commercially available controllers.

andytheengineerguy, 2024-01-16:
@wata-net ok, no problem. I just bought a Nucleo. I'll let you know how it goes

IGR2014, 2024-09-13:
@wata-net I can do it if you explain me how SM16169 should be initialized. I see your code, it's ok but without some kind of programming manual it's hard to convert this code for rpi-rgb-led-matrix library

wata-net, 2024-09-16:
@IGR2014 I'm sorry, but I don't have enough time to create a manual yet.
I've posted before about how to initialize SM16169, but if you have any questions, could you let me know?
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1504#issuecomment-1406583087
I'll answer as much as I know.

IGR2014, 2024-09-17:
Oh, wow, thank you! I missed your screenshot with register map. I'll try to make working code ASAP

wata-net, 2024-09-21:
@IGR2014 
I don't know the details.
It's probably some kind of unlock command to write to the configuration registers.

Endorfin35, 2025-02-26:
@wata-net , How did you access the register map information for SM16169? If you have a datasheet, could you share it?

wata-net, 2025-02-26:
@Endorfin35, Since there is no datasheet available, I analyzed the signals and created a register map.

wata-net, 2025-02-28:
@Endorfin35 ,

8K of SRAM can be calculated as follows.
Grayscale resolution x number of outputs x number of scan lines x double buffer, so 2
Grayscale resolution = 16 bits
Number of outputs = 16 outputs
Number of scan lines supported by SM16169 = 16
In other words, 16*16*16*2 = 8192 bits.

I don't remember the detailed procedure of data transmission very well.
The timing of receiving grayscale data and the timing of scanning the LED panel are probably completely independent for the driver IC with built-in SRAM.
The R, G, and B data are fed into the SRAM of the driver IC with DCLK and LAT.
This timing can be arbitrary and does not matter. For video playback applications, we recommend transferring at the same speed as the video frame rate (60 FPS, 30 FPS, or 23.976 FPS, etc.).
When the entire screen's worth of data has been transferred, the driver IC automatically switches to the other side of the double buffer.
If you just keep sending data continuously, it seems like you won't know where the data is written in the SRAM, but that's why there's the vertical synchronization command.
After transferring one screen's worth of data, send the vertical synchronization command before sending the data from the beginning again to match the start of the data to be transferred with the start of the SRAM in the driver IC.
I think there was a rule that the GCLK clock transmission had to be completed before sending the vertical synchronization command.
The timing of the refresh rate is determined by the A, B, C (LCK, BK, DIN) signals that switch the LED lines and the clock input to GCLK.
There are various technologies to improve the visual refresh rate, such as SM-PWM and Scrambled PWM, but in any case, inputting a specified number of clocks to GCLK will complete the display of one LED line.
Input signals to LCK, BK, and DIN to advance the LED by one line. Input the specified clock to GCLK again to display one LED line.
When the entire screen has been displayed, it will display again from the first line.
At this time, the data to be displayed on the LED uses a buffer that is not currently being written.

I think it was something like this.

Endorfin35, 2025-02-28:
This is very valuable information. Thank you for the explanation. I was calculating and staying at 4096 instead of 8912. If there is a double buffer, the calculation is 8192. so,  When the outputs are active, I write the data to the unused buffer and then should I send v sync to activate/replace the buffer? Do I understand correctly?

wata-net, 2025-02-28:
@Endorfin35 ,
I think that's probably it.
I'm just guessing at the mechanism based on the results of analysis and trial and error, so I don't think it's accurate.

YassineElHosni, 2025-04-30:
any updates on this?

marcmerlin, 2025-04-30:
@YassineElHosni those panels are and have always been incompatible with this library.
There is no support being written for them unless you were planning on writing it yourself or paying someone to do it for you.
But realistically PWM panels are very different and would require a very different driver, even if it would be a lot simpler once fully written.

YassineElHosni, 2025-04-30:
thank you for your time @marcmerlin

YassineElHosni, 2025-05-01:
@marcmerlin a quick question, do we have a list of supported chips, my supplier is asking, and i cant seem to find something about this, please help 🙏🏻

marcmerlin, 2025-05-01:
@YassineElHosni I feel someone was discussing this on the discourse group, but I can't find the thread. Maybe if you search you can find it there or in a bug here.
More generally, yes, there should be a list of supported chips in the github README and currently there is not.  I will happily accept git contribution to add a page with all supported chips as well as those known not working.
In the meantime, you can have a look at
https://rpi-rgb-led-matrix.discourse.group/t/panel-recommendations/927/6
and
https://rpi-rgb-led-matrix.discourse.group/t/p10rgb-3535-4s-zaggiz-1-4scan-abc-p10-panels-and-https-github-com-2dom-pxmatrix/921

board707, 2025-05-01:
> there should be a list of supported chips 

Perhaps this tables will helps:
https://github.com/board707/DMD_STM32/wiki/Led_drivers

Panel main led driver chips are in the second table. Every driver with not a  PWM type, that labelled with green "Tested" icon should be compatible with the hzeller library.

marcmerlin, 2025-05-01:
Thanks @board707  for now, I added this section in the wiki:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md#panels-supported

board707, 2025-05-01:
I think you have to add some remarks about PWM type chips, because the table contains some of them, labelled as "tested". 
Nevertheless, none of such drivers are compatible with the current hzeller code.

marcmerlin, 2025-05-01:
@board707  did you read the new page? :)  first thing it says is no PWM panel is supported

marcmerlin, 2025-05-01:
@board707  but I do welcome pull requests, please send your updates to that section. You don't even need to use git, you can just go on github, edit the README on the web and send a pull request without typing a single git command :)

board707, 2025-05-01:
Honestly I didn't. I am in a hotel now typing this on a phone with a single finger:)

2 мая 2025 г. 00:52:13 Marc MERLIN ***@***.***>:

> did you read the new page? :)

YassineElHosni, 2025-05-02:
Thank you for sharing, much appreciated @marcmerlin @board707

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin