hzeller/rpi-rgb-led-matrix #1507: Flask App question

galaxyZ2, 2023-01-31:
Hello Everyone,

I am totally new to programming, and I choose a hobby project in the form of a raspberry pi 4 with an adafruit-hat controlling a 32x32 matrix.

I wrote a python script, which is working quite beautifully, however when I use the text mover functions (either of them) the text I initially set will change to "utf-8" and keep moving according to the function.

What could cause this?
The code is from a beginner, and is not finished. :)

Any help is much appreciated!

`from flask import Flask, request, render_template, redirect
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
from flask_wtf import FlaskForm
from wtforms import StringField, SubmitField, SelectField
import logging, os
from pprint import pprint

app = Flask(__name__)
app.config['SECRET_KEY'] = '19901009'

# Initialize and configure LED matrix
options = RGBMatrixOptions()
options.rows = 32
options.cols = 32
options.chain_length = 1
options.brightness = 100
options.pwm_bits = 1
options.pwm_lsb_nanoseconds = 100
options.gpio_slowdown = 5
options.hardware_mapping = 'adafruit-hat'
matrix = RGBMatrix(options = options)

# Create a canvas to draw on
canvas = matrix.CreateFrameCanvas()
x = int
y = int
message = str
color = str

#Load default font.
font = graphics.Font()

#Read available fonts into variable

class DisplayForm(FlaskForm):
    message = StringField('Messages')
    color = StringField('Color')
    submit = SubmitField('Submit')
    
@app.route('/display', methods=['GET','POST'])

def display():
    global color, x, y
    #logging.basicConfig(level=logging.DEBUG, format='%(asctime)s %(message)s', datefmt='%m/%d/%Y %I:%M:%S %p')
    form = DisplayForm()
    if form.validate_on_submit():
        if not form.message.data or not form.color.data:
           return render_template('display.html', form=form, error_mesage='Please enter the values')
        x = 0
        y = 22 
        message = form.message.data
        color_code = form.color.data
        r, g, b = map(int, color_code.split(","))
        color = graphics.Color(r,g,b)
        font.LoadFont('/home/linux/rpi-rgb-led-matrix/fonts/7x13.bdf')
        canvas.Clear()
        graphics.DrawText(canvas, font, x, y, color, message)
        matrix.SwapOnVSync(canvas)
        #return 'Displaying message: {}'.format(message)
    return render_template('display.html', form=form, color=color, x=x, y=y)
    
@app.route('/stop', methods=['GET', 'POST'])
def stop():
    form = DisplayForm()
    #Clear the canvas
    canvas.Clear()
    #Push the cleared canvas to the matrix
    matrix.SwapOnVSync(canvas)
    return render_template('display.html', form=form, x=x, y=y)

@app.route('/move_left', methods=['GET', 'POST'])
def move_left():
    global x, message
    x = x - 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_right', methods=['GET', 'POST'])
def move_right():
    global x 
    x = x + 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_up', methods=['GET', 'POST'])
def move_up():
    global y
    y -= 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

@app.route('/move_down', methods=['GET', 'POST'])
def move_down():
    global y, message 
    y += 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

if __name__ == '__main__':
    app.run(host='192.168.1.112', port=1234)`

galaxyZ2, 2023-01-31:
`from flask import Flask, request, render_template, redirect
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
from flask_wtf import FlaskForm
from wtforms import StringField, SubmitField, SelectField
import logging, os
from pprint import pprint

app = Flask(__name__)
app.config['SECRET_KEY'] = '19901009'

# Initialize and configure LED matrix
options = RGBMatrixOptions()
options.rows = 32
options.cols = 32
options.chain_length = 1
options.brightness = 100
options.pwm_bits = 1
options.pwm_lsb_nanoseconds = 100
options.gpio_slowdown = 5
options.hardware_mapping = 'adafruit-hat'
matrix = RGBMatrix(options = options)

# Create a canvas to draw on
canvas = matrix.CreateFrameCanvas()
x = int
y = int
message = str
color = str

#Load default font.
font = graphics.Font()

#Read available fonts into variable

class DisplayForm(FlaskForm):
    message = StringField('Messages')
    color = StringField('Color')
    submit = SubmitField('Submit')
    
@app.route('/display', methods=['GET','POST'])

def display():
    global color, x, y
    #logging.basicConfig(level=logging.DEBUG, format='%(asctime)s %(message)s', datefmt='%m/%d/%Y %I:%M:%S %p')
    form = DisplayForm()
    if form.validate_on_submit():
        if not form.message.data or not form.color.data:
           return render_template('display.html', form=form, error_mesage='Please enter the values')
        x = 0
        y = 22 
        message = form.message.data
        color_code = form.color.data
        r, g, b = map(int, color_code.split(","))
        color = graphics.Color(r,g,b)
        font.LoadFont('/home/linux/rpi-rgb-led-matrix/fonts/7x13.bdf')
        canvas.Clear()
        graphics.DrawText(canvas, font, x, y, color, message)
        matrix.SwapOnVSync(canvas)
        #return 'Displaying message: {}'.format(message)
    return render_template('display.html', form=form, color=color, x=x, y=y)
    
@app.route('/stop', methods=['GET', 'POST'])
def stop():
    form = DisplayForm()
    #Clear the canvas
    canvas.Clear()
    #Push the cleared canvas to the matrix
    matrix.SwapOnVSync(canvas)
    return render_template('display.html', form=form, x=x, y=y)

@app.route('/move_left', methods=['GET', 'POST'])
def move_left():
    global x, message
    x = x - 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_right', methods=['GET', 'POST'])
def move_right():
    global x 
    x = x + 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_up', methods=['GET', 'POST'])
def move_up():
    global y
    y -= 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

@app.route('/move_down', methods=['GET', 'POST'])
def move_down():
    global y, message 
    y += 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

if __name__ == '__main__':
    app.run(host='192.168.1.112', port=1234)`

galaxyZ2, 2023-01-31:
'''
from flask import Flask, request, render_template, redirect
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
from flask_wtf import FlaskForm
from wtforms import StringField, SubmitField, SelectField
import logging, os
from pprint import pprint

app = Flask(__name__)
app.config['SECRET_KEY'] = '19901009'

# Initialize and configure LED matrix
options = RGBMatrixOptions()
options.rows = 32
options.cols = 32
options.chain_length = 1
options.brightness = 100
options.pwm_bits = 1
options.pwm_lsb_nanoseconds = 100
options.gpio_slowdown = 5
options.hardware_mapping = 'adafruit-hat'
matrix = RGBMatrix(options = options)

# Create a canvas to draw on
canvas = matrix.CreateFrameCanvas()
x = int
y = int
message = str
color = str

#Load default font.
font = graphics.Font()

#Read available fonts into variable

class DisplayForm(FlaskForm):
    message = StringField('Messages')
    color = StringField('Color')
    submit = SubmitField('Submit')
    
@app.route('/display', methods=['GET','POST'])

def display():
    global color, x, y
    #logging.basicConfig(level=logging.DEBUG, format='%(asctime)s %(message)s', datefmt='%m/%d/%Y %I:%M:%S %p')
    form = DisplayForm()
    if form.validate_on_submit():
        if not form.message.data or not form.color.data:
           return render_template('display.html', form=form, error_mesage='Please enter the values')
        x = 0
        y = 22 
        message = form.message.data
        color_code = form.color.data
        r, g, b = map(int, color_code.split(","))
        color = graphics.Color(r,g,b)
        font.LoadFont('/home/linux/rpi-rgb-led-matrix/fonts/7x13.bdf')
        canvas.Clear()
        graphics.DrawText(canvas, font, x, y, color, message)
        matrix.SwapOnVSync(canvas)
        #return 'Displaying message: {}'.format(message)
    return render_template('display.html', form=form, color=color, x=x, y=y)
    
@app.route('/stop', methods=['GET', 'POST'])
def stop():
    form = DisplayForm()
    #Clear the canvas
    canvas.Clear()
    #Push the cleared canvas to the matrix
    matrix.SwapOnVSync(canvas)
    return render_template('display.html', form=form, x=x, y=y)

@app.route('/move_left', methods=['GET', 'POST'])
def move_left():
    global x, message
    x = x - 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_right', methods=['GET', 'POST'])
def move_right():
    global x 
    x = x + 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_up', methods=['GET', 'POST'])
def move_up():
    global y
    y -= 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

@app.route('/move_down', methods=['GET', 'POST'])
def move_down():
    global y, message 
    y += 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

if __name__ == '__main__':
    app.run(host='192.168.1.112', port=1234)'''

galaxyZ2, 2023-01-31:
Apologies, I dont know how to comment the code in a readable format:

`from flask import Flask, request, render_template, redirect
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
from flask_wtf import FlaskForm
from wtforms import StringField, SubmitField, SelectField
import logging, os
from pprint import pprint

app = Flask(__name__)
app.config['SECRET_KEY'] = '19901009'

# Initialize and configure LED matrix
options = RGBMatrixOptions()
options.rows = 32
options.cols = 32
options.chain_length = 1
options.brightness = 100
options.pwm_bits = 1
options.pwm_lsb_nanoseconds = 100
options.gpio_slowdown = 5
options.hardware_mapping = 'adafruit-hat'
matrix = RGBMatrix(options = options)

# Create a canvas to draw on
canvas = matrix.CreateFrameCanvas()
x = int
y = int
message = str
color = str

#Load default font.
font = graphics.Font()

#Read available fonts into variable

class DisplayForm(FlaskForm):
    message = StringField('Messages')
    color = StringField('Color')
    submit = SubmitField('Submit')
    
@app.route('/display', methods=['GET','POST'])

def display():
    global color, x, y
    #logging.basicConfig(level=logging.DEBUG, format='%(asctime)s %(message)s', datefmt='%m/%d/%Y %I:%M:%S %p')
    form = DisplayForm()
    if form.validate_on_submit():
        if not form.message.data or not form.color.data:
           return render_template('display.html', form=form, error_mesage='Please enter the values')
        x = 0
        y = 22 
        message = form.message.data
        color_code = form.color.data
        r, g, b = map(int, color_code.split(","))
        color = graphics.Color(r,g,b)
        font.LoadFont('/home/linux/rpi-rgb-led-matrix/fonts/7x13.bdf')
        canvas.Clear()
        graphics.DrawText(canvas, font, x, y, color, message)
        matrix.SwapOnVSync(canvas)
        #return 'Displaying message: {}'.format(message)
    return render_template('display.html', form=form, color=color, x=x, y=y)
    
@app.route('/stop', methods=['GET', 'POST'])
def stop():
    form = DisplayForm()
    #Clear the canvas
    canvas.Clear()
    #Push the cleared canvas to the matrix
    matrix.SwapOnVSync(canvas)
    return render_template('display.html', form=form, x=x, y=y)

@app.route('/move_left', methods=['GET', 'POST'])
def move_left():
    global x, message
    x = x - 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_right', methods=['GET', 'POST'])
def move_right():
    global x 
    x = x + 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')
    
@app.route('/move_up', methods=['GET', 'POST'])
def move_up():
    global y
    y -= 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

@app.route('/move_down', methods=['GET', 'POST'])
def move_down():
    global y, message 
    y += 1
    canvas.Clear()
    graphics.DrawText(canvas, font, x, y, color, message)
    matrix.SwapOnVSync(canvas)
    return redirect('/')

if __name__ == '__main__':
    app.run(host='192.168.1.112', port=1234)`

hzeller, 2023-01-31:
Usually you'd do triple ticks to have code shown properly formatted.

It looks like you're not assigning the return value of `SwapToVSync()` to canvas. Without that, you don't have double-buffering.

```Python
  canvas = matrix.SwapOnVSync(canvas)
```

ledvinap, 2023-02-01:
(can you repost formated code / edit post? there is 'add code' icon above post form)

My guess is that `message` variable in `display()` is local, other functions refer `message` from global scope

galaxyZ2, 2023-02-01:
> (can you repost formated code / edit post? there is 'add code' icon above post form)
> 
> My guess is that `message` variable in `display()` is local, other functions refer `message` from global scope

BINGO!! Thank you! What a small mistake, but it works beautifully if i change it! Thank you!

galaxyZ2, 2023-02-01:
> Usually you'd do triple ticks to have code shown properly formatted.
> 
> It looks like you're not assigning the return value of `SwapToVSync()` to canvas. Without that, you don't have double-buffering.
> 
> ```python
>   canvas = matrix.SwapOnVSync(canvas)
> ```

Thank you! It ended up being the second comment! All good thank you for the ideas!

hzeller, 2023-02-01:
You still need `canvas = matrix.SwapOnVSync(canvas)` for a flickering-free image transition.

galaxyZ2, 2023-02-01:
> You still need `canvas = matrix.SwapOnVSync(canvas)` for a flickering-free image transition.

Forgot to mention that I did add the line as suggested! Much appreciated!

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin