hzeller/rpi-rgb-led-matrix #1508: Font loading

galaxyZ2, 2023-02-02:
Hello Everyone!

Please allow me another question I am fighting for quite a bit.
I have a flask app, and the webpage prompts the user to choose a font type.

`#Read available fonts into variable
fontname = []
font_dir = '/home/linux/rpi-rgb-led-matrix/rpi-rgb-led-matrix/fonts/'
for root, dirs, files in os.walk(font_dir):
    for file in files:
        if file.endswith(".bdf"):
            fontname.append(file) 
`

galaxyZ2, 2023-02-02:
I am not sure if in the graphics.DrawText() font needs to be mentioned or font_path, neither of them have possitive effect on the outcome.

galaxyZ2, 2023-02-02:
> font_path = graphics.Font()

I am now trying to create a font variable as follows then load that, but to no avail. :(

```
 font_path = os.path.join(font_dir, fonttype)
 font = graphics.Font(font_path)
 font.LoadFont(font)
 graphics.DrawText(canvas, font, x, y, color, message)
```

ledvinap, 2023-02-02:
font_path = os.path.join(font_dir, fonttype)  # string with path
 font = graphics.Font();   # empty Font object
 font.LoadFont(font_path ) # load font file into font object
 graphics.DrawText(canvas, font, x, y, color, message)

galaxyZ2, 2023-02-02:
> 

Got it working, thank you for your help! I also had bad reference in the html.

closed 2025-11-27 by marcmerlin