hzeller/rpi-rgb-led-matrix #1509: size_t negative value

galaxyZ2, 2023-02-03:
Hello Everyone!

I am at the very last step to finish my little hobby project.

I am uploading a picture to my flask /upload function from image.html:

```
<div class="container">
      <h1 class="text-center">LED Matrix Display</h1>
	<form action="/upload" method="post" enctype="multipart/form-data">
		<input type="file" id="upload-file" name="image">
		<label for="upload-file">Choose a file</label>
		<input type="submit" value="Upload">
	</form>
	</div>
```

My functions looks like this:

```
@app.route("/upload", methods=['POST'])
def upload():
    image = request.files['image']
    image = Image.open(image)
    matrix = RGBMatrix(options = options)
    image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
    matrix.SetImage(image.convert('RGB'))
    return render_template("image.html")
```

The console returns, OverflowError: can't convert negative value to size_t

I have used a PNG 32x32 pixels to ensure it fits regardless.

I have been looking into core.pyx where the issue is mentioned (l22 and l45)

After many hours I am clueless. Could one of you please help out to finish my project? :)

have a great day everyone and thanks!

ledvinap, 2023-02-03:
Can you (at least) post full backtrace?

galaxyZ2, 2023-02-03:
> Can you (at least) post full backtrace?

Sure sorry, here we go :

```
^Croot@raspberrypi:/home/linux/rpi-rgb-led-matrix/rpi-rgb-led-matrix/bindings/python/samples# python3 server.py
Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
 * Serving Flask app "server" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://192.168.1.112:1234/ (Press CTRL+C to quit)
Can't set realtime thread priority=99: Operation not permitted.
        You are probably not running as root ?
        This will seriously mess with color stability and flicker
        of the matrix. Please run as `root` (e.g. by invoking this
        program with `sudo`), or setting the capability on this
        binary by calling
        sudo setcap 'cap_sys_nice=eip' /usr/bin/python3.9
[2023-02-03 14:17:30,485] ERROR in app: Exception on /upload [POST]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/linux/rpi-rgb-led-matrix/rpi-rgb-led-matrix/bindings/python/samples/server.py", line 173, in upload
    matrix.SetImage(image.convert('RGB'))
  File "core.pyx", line 22, in rgbmatrix.core.Canvas.SetImage
  File "core.pyx", line 46, in rgbmatrix.core.Canvas.SetPixelsPillow
OverflowError: can't convert negative value to size_t
192.168.1.101 - - [03/Feb/2023 14:17:30] "POST /upload HTTP/1.1" 500 -
```

ledvinap, 2023-02-03:
This may be bug in https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/rgbmatrix/core.pyx#L45-L46. Pillow returns pointer as `Py_ssize_t`, but `uintptr_t` cast is used. Possible fix (from https://cython.readthedocs.io/en/latest/src/userguide/language_basics.html#id27) 
```
cdef extern from *:
    ctypedef Py_ssize_t Py_intptr_t
...

image_ptr = (<uint32_t **>(<Py_intptr_t>ptr_tmp))

```
But is is strange, I'd expect userspace memory below 0x8000000 .. 

One possible workaround is to use `matrix.SetImage(image.convert('RGB'), unsafe=False)`, with some performance penalty.

galaxyZ2, 2023-02-03:
> This may be bug in https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/rgbmatrix/core.pyx#L45-L46. Pillow returns pointer as `Py_ssize_t`, but `uintptr_t` cast is used. [...]

You are a legend! unsafe=False did it! I only need to figure out why it is only using red colors. 

THanks a lot!

hzeller, 2023-02-03:
Question to the Python folks: is there a way we can work around it in the python bindings code, e.g. by adding a type-cast at the right place or use a different type ?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin