hzeller/rpi-rgb-led-matrix #1510: Red leds don't work on the top half

psabato1987, 2023-02-07:
Hi all
I'm working with a RPi 3 Model A+ running Raspberry Pi OS Lite, an Adafruit RGB Matrix Bonnet and a RGB 64x64 matrix.
I melt a blob of solder between E and 8, run the rgb-matrix.sh script with the “convenience” setting.
i2c and 1-wire disabled.
Then I run the command sudo ./demo -D 0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=4 --led-gpio-mapping=adafruit-hat but the red leds in the top half are not working.

https://imgur.com/a/SOXzs9J

Could you please help me with this issue?
Thank you

psabato1987, 2023-02-07:
ISSUE FIXED

I tried to set these parameters to 0 in /boot/config.txt

camera_auto_detect=0
display_auto_detect=0

I don't know if the issue is related to these parameters but in the meantime I realised that the Adafruit Bonnet is pushed up by the USB port of the Raspberry Pi3 model A.
I tried to push down the bonnet and it worked fine for a few moments.
So maybe this is the reason why the issue occurs whit this Pi.
I will try a 40 Pin Stacking Female Header Kit in order to avoid this.

HawtDogFlvrWtr, 2023-02-20:
I'm having the same issue with mine, but i don't have a bonnet. It's wired directly to the gpio's of the pi. DId you ever figure out exactly what it was? I changed your two settings with no result

HawtDogFlvrWtr, 2023-02-20:
never mind. i had the wiring wrong :D

PhilBReed, 2023-03-02:
Stacking header did the trick for me.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828