hzeller/rpi-rgb-led-matrix #1511: Display both contenus reduce brightness

romsera, 2023-02-07:
Hi all,

I am trying to display both text and image supperposed in same time but it  reduce led brightness. It seams the two medias are displayed alternatively.

How to conserve the maximum of brightness when display several contents at same time ?

Is it possible to create a combined stream that contains all supperposed contents ?

Thx.

Romain.

hzeller, 2023-02-07:
Do you try to run two binaries such as the text viewer and image viewer at once ? That of course does not work as they will fight the output GPIO.

You need to write a single  program that shows all the content you need. Or use something like a networked compositor like flaschen-taschen and send data from there.

romsera, 2023-02-07:
Thank you for your quicly response.

indeed I cumulated the calls and I understood the inappropriate behavior.

Isn't there a possibility to merge several feeds before displaying them?

I will also test your solution.

Thank you again for your answer.

Regards,

Romain.

ledvinap, 2023-02-08:
Hard to tell without code.
Maybe using double-buffering will help you - compose into offscreen canvas and then display it: `offscreen = matrix.SwapOnVSync(offscreen)`

romsera, 2023-04-28:
I try to follow your advice by using the python binding of your library and drawing my content through PILLOW before swapping it to the display.

The layout of my led tiles is not rectangular but the content I want to display is. Which means that some pixels will not be visible (cropped). The idea is therefore to "consume" these non-visible pixels directly in the SetImage and SwapOnVsync functions.

I find the SetImage function in the core.pyx file but I can't find SwapOnVsync. Where should I modify this function?

Thx.

ledvinap, 2023-04-28:
THis may help you: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/image-scroller.py#L17-L30

romsera, 2023-04-28:
Thx for this link but not exactly the same usage and not realy helping me.

The main goal is to avoid the display of pixels where no panel existing.

Example of square image to display :

| 1 | 2 | 3 |
| 4 | 5 | 6 |
| 7 | 8 | 9 |

The layout of my panels :

     | A |
| B | C | D |
     | E |

Actual result :

     | 1 |
| 4 | 5 | 6 |
     | 7 |

Expected result after consume pixel of image parts 1, 3, 7 and 9 :

     | 2 |
| 4 | 5 | 6 |
     | 8 |

Hope it is more clearly with that schems :)

ledvinap, 2023-04-28:
You can try https://github.com/hzeller/rpi-rgb-led-matrix/pull/1478, marking corner panels as 'discard'. The canvas will be 3x3 square (and you can use chains of length 2, improving performance a bit).

Or you can use pillow to move panel pixels (2->1, 8>7) and then use upstream version.

ledvinap, 2023-05-03:
You need to compile my branch with Remap support. It is not supported upstream.

In Remap fragment, you should specify panel orientation (NSEW). In your case, it is probably West for all panels. If you want to use shorted cables, you can alternate West and East panels.

The remap line is list for each chain (including missing panels at the end of chain), concatenated. So first entry is position of first panel on chain 1 (panel 1), at position 32, 0. Seventh entry is position of first panel in chain 2 (panel 3).

It should be something like 
```
Remap:96,96|32,0w|48,0w|0,0x|0,0x|0,0x|0,0x|0,32w|16,32w|32,32w|48,32w|64,32w|80,32w|32,64w|48,64w|0,0x|0,0x|0,0x|0,0x;
```
(not 100% sure I got it correct)

Also you don't need V-mapper and Rotate, just place panels correctly (including orientation). Only mirror is not implemented in Remap (and probably only needed in obscure cases). Use sketch of panels in final position and orientation and then write remap fragments with position from image, following chains.

When this arrangement is working, you can improve performance by rearranging cabling so that each chain is 3 or 4 panels long (30% improvement).

romsera, 2023-05-04:
Thx a lot for your help but after compiling your branch I still blocked with the same "Remap: no such mapper" error.

Discard panels must be specified at the end of each line or may they be specified like this :

```
options.pixel_mapper_config = "Remap:96,96|0,0x|16,0x|32,0w|48,0w|64,0x|80,0x|0,32w|16,32w|32,32w|48,32w|64,32w|80,32w|96,32w|0,48x|16,48x|32,48w|48,48w|64,48x|80,48x"
```

ledvinap, 2023-05-04:
Can you try to run some example?
```
./demo --led-rows=16 --led-chain=6 --led-parallel=3 --led-pixel-mapper='Remap:96,96|32,0w|48,0w|0,0x|0,0x|0,0x|0,0x|0,32w|16,32w|32,32w|48,32w|64,32w|80,32w|32,64w|48,64w|0,0x|0,0x|0,0x|0,0x' -D1 runtext.ppm
```

It seems to work for me (but I can't test with panel now)

romsera, 2023-05-04:
OK I anderstand.

This cmd works :
```
sudo ./examples-api-use/demo --led-rows=16 --led-cols=32 --led-chain=6 --led-parallel=3 --led-pixel-mapper='Remap:96,96|32,0e|48,0e|0,0x|0,0x|0,0x|0,0x|0,32e|16,32e|32,32e|48,32e|64,32e|80,32e|32,64e|48,64e|0,0x|0,0x|0,0x|0,0x' --led-multiplexing=18 --led-row-addr-type=2 --led-brightness=30 -D0
```

This one too but display scrolling text only on top panel :
```
sudo ./examples-api-use/demo --led-rows=16 --led-cols=32 --led-chain=6 --led-parallel=3 --led-pixel-mapper='Remap:96,96|32,0e|48,0e|0,0x|0,0x|0,0x|0,0x|0,32e|16,32e|32,32e|48,32e|64,32e|80,32e|32,64e|48,64e|0,0x|0,0x|0,0x|0,0x' --led-multiplexing=18 --led-row-addr-type=2 --led-brightness=30 -D1 ./examples-api-use/runtext.ppm
```

Now, other issue :( two white lines are displayed verticaly on the bottom of my display :

[image]

Is this due to to mapping ?

ledvinap, 2023-05-04:
You should try some other demo (maybe  ./text-example ) to get better idea about canvas-to-display mapping.

Or use larger image (height of 96 pixels, width at least 96 pixels) with demo. Try some test image where it is obvious what part of image is displayed (3x3 grid with numbers + marks).

Or try `pixel-mover` example

ledvinap, 2023-05-04:
> Now, other issue :( two white lines are displayed verticaly on the bottom of my display

Is bottom part OK without mapper?

romsera, 2023-05-04:
Others demos work fine. For example the cube is correctly displayed :

[image]

Seems two halfs of two bottom panels are displaying white continuously.

romsera, 2023-05-05:
I can't try to swap out the ribbon cables until next Tuesday.

However, I don't think that's it because if I don't use a mapper and configure the broadcast with V-mapper and 90 degree rotation, a manually swapped pillow image displays correctly.

ledvinap, 2023-05-05:
I'll try to test if .. 

Can you post picture without mapper, please?

romsera, 2023-05-05:
OK, I cannot test today because no hardware with me.

I will test and send you picture on next Tuesday (Monday is public holliday in france)

thx for all.

romsera, 2023-05-09:
> 

swapping ribbon cables just move the problem :

[image]

Result with the same image and without mapping :

[image]

The configuration : 

```python
options.rows = 16
options.cols = 32
options.chain_length = 6
options.parallel = 3
options.pwm_bits = 8
options.pwm_lsb_nanoseconds = 300
options.row_address_type = 2
options.multiplexing = 18
options.brightness = 20
```

ledvinap, 2023-05-09:
The white strips look like cable (unlikely), adapter or IO problem. 
(R1)G1B1 (or (R2)G2B2, I'm not sure) on strip 3 are fixed to logical 1 for some reason. 
You can try swapping chain2 <-> chain3, it should result in stripes on all three panels.

Is white really displayed in stripes? Or is it cyan? In that case, number 5 shall be partially visible in stripe (not sure from photo). That would mean that only G2B2 or G1B1 is fixed.

G1B1 is mapped to I2C1 pins. Maybe you have I2C1 enabled?

romsera, 2023-05-09:
Yes, if I swap chain2 with chain 3, stripes are displayed on all three panels.

Stipes are displayed in cyan and not in white.

I forgot to mention it but I2C is used for RTC clock and the use of two sensors.

ledvinap, 2023-05-09:
It's either chain3 or I2C1 ... 
It may be possible to use I2C0.
Or rearrange panels into 2 chains. With Remap, you can use any combination, so select variant with short, simple cabling (you can even rotate individual panels) and then write remap parameters ...

closed 2025-11-27 by marcmerlin