hzeller/rpi-rgb-led-matrix #1515: Trying to port lib to work with rk3566 and rk3399.

kvjoshi, 2023-02-17:
Due to the recent shortages of rpi , im trying to make the script work on rk3599 initially and later on rk3399 for rock 3 and rockpi 4.

any guidance would be appreciated if anyone has tried to do so.

hzeller, 2023-02-17:
You essentially need to read the datasheet of your SoC and re-implement the interface provided by [gpio.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.h). If your SoC has a way to memory map GPIO pins (most important for fast access), you should be ready to go.

If you also has a way to create nicely timed pulses per hardware (as implemented as the `HardwarePinPulser` in gpio.cc, the output might also be pretty glitch-free. If not, the fallback implementation that manually wiggles the pin with a `usleep()` is a good default.

kvjoshi, 2023-02-18:
ok thanks , looking at the datasheets now.

652436962, 2023-04-04:
> ok thanks , looking at the datasheets now.

hello  have you  compulish  it?

xuniuer, 2023-05-04:
Eager for further update on porting lib with rock chip...

jaygarcia, 2023-06-25:
> You essentially need to read the datasheet of your SoC and re-implement the interface provided by [gpio.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.h). If your SoC has a [...]

Thank you @hzeller ! I started to look at the internals to begin porting this for the Orange Pi5B. Assuming i can get one set working with being wired directly, I will work on a compatible daughter board using the schematics you have in this repo as a base. 

My goal is to create a PR for all of this to use and hopefully get Electro Dragon to mass produce the daughter boards. 

More to come :)

kvjoshi, 2023-06-30:
@jaygarcia all the best. due to changes in my schedule, I had to bench the port for rockchip.

SturdyCat, 2023-10-15:
> no. i put it on hold , mapping different gpio blocks is a little time consuming. shifted to teensy SmartMatrix as an alternative. will be resuming next month, was facing issues with accessing dev/mem

@kvjoshi  Closing kernel "CONFIG_STRICT_DEVMEM" makes "mmap" work.

phongans, 2023-12-22:
@SturdyCat , How to shutdown kernel CONFIG_STRICT_DEVMEM=n

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin