hzeller/rpi-rgb-led-matrix #1523: Wiring ...

targettadams, 2023-03-16:
Hi,

Thank you for this truly awesome resource you've put together. This isn't exactly an issue to report, but I couldn't see another way of communicating with you. I've been studying your wiring/connection set-up as depicted here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/powerbar.jpg. 

I have a similar-ish set-up which I wish to implement. Could I just ask - did you power the raspberry pi off the same 5V supply? I have a set-up consisting of a pi 3B (requires its own power), adafruit matrix bonnet (requires its own power), and 6 panels. I would like the user to have just one plug to power the device, as a whole, rather than 3. I'm a little unsure of how to wire up the pi (and the adafruit for that matter) to the power supply (my coding is decent, my electronics not so good :)). Was wondering if you had some pointers?

Thanks in advance

combs, 2023-05-08:
suggest the adafruit forums/discord for this question since it's more of an adafruit bonnet power supply question...

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin