hzeller/rpi-rgb-led-matrix #1525: No image from python; 64x64 matrix and bonnet

joeleehen, 2023-03-26:
Howdy, fellers. I'm stuck at a bit of a crossroads with my project. I'm using a 64x64 matrix, a raspberry pi 0, and an adafruit bonnet. I'm pretty confident I've done the required hardware setup and I'm the latest version of this repository.
I get a very stable image from the matrix when I use the C and python examples, everything functions as expected. Changing the hardware mapping to `adafruit-hat` makes the image flickery and unstable.
When I set the image with my own python code, however, the image looks just like the demo images did when using the HAT hardware mapping. When I explicitly set the matrix options to use the bonnet (i.e. `options.hardware_mapping='regular'`) I get no image from the matrix. It's almost as if it defaults to using the HAT.
Has anyone else had this problem?

joeleehen, 2023-04-04:
I stepped away and came back after a while and it seems the issue has solved itself? The image quality has drastically improved and barely flickers. I think the image still occasionally flickers like it's using the wrong hardware mapping but a quick reboot usually fixes it.

closed 2023-04-04 by joeleehen

hzeller, 2023-04-04:
For low flcker, make sure you have the [improve flicker](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) hardware mod needed for the Adafruit hat/bonnet.