hzeller/rpi-rgb-led-matrix #1526: fatal error: longintrepr.h: No such file or directory

ceilingduster, 2023-03-27:
Running into the following error.  It seems to be related to using a newer version of Python (3.11.2 vs 3.9).  

```
...
make[1]: Leaving directory '/home/iredden/rpi-rgb-led-matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/iredden/rpi-rgb-led-matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/iredden/rpi-rgb-led-matrix/lib'
make[2]: Leaving directory '/home/iredden/rpi-rgb-led-matrix/lib'
/usr/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext
building 'core' extension
creating build
creating build/temp.linux-armv7l-cpython-311
creating build/temp.linux-armv7l-cpython-311/rgbmatrix
gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I../../include -I/usr/local/include/python3.11 -c rgbmatrix/core.cpp -o build/temp.linux-armv7l-cpython-311/rgbmatrix/core.o -O3 -Wall
rgbmatrix/core.cpp:198:12: fatal error: longintrepr.h: No such file or directory
  198 |   #include "longintrepr.h"
      |            ^~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1
make[1]: *** [Makefile:67: build-python] Error 1
make[1]: Leaving directory '/home/iredden/rpi-rgb-led-matrix/bindings/python'
make: *** [Makefile:28: build-python] Error 2
Configuring system...
Done.
```

Any ideas how to resolve this?  I require to use Python 3.11.2.

ledvinap, 2023-03-27:
Running `cython` to generate binding files may help (`python3 -m cython -a -3 --cplus *.pyx`)

dogerber, 2023-10-20:
I have the same problem when trying to build

```
make build-python PYTHON=$(command -v python3)
```

in the bindings/python folder

dogerber, 2023-10-21:
> Running `cython` to generate binding files may help (`python3 -m cython -a -3 --cplus *.pyx`)

I tried this but nothing changed from it.

WLedden, 2023-10-25:
> > Running `cython` to generate binding files may help (`python3 -m cython -a -3 --cplus *.pyx`)
> 
> I tried this but nothing changed from it.

Have you found how to fix the issue yet?

dogerber, 2023-10-26:
> > > Running `cython` to generate binding files may help (`python3 -m cython -a -3 --cplus *.pyx`)
> > 
> > 
> > I tried this but nothing changed from it.
> 
> Have you found how to fix the issue yet?

Unfortunately not. I tried using an environment with Python3.9 (as that seemed to work for others previously) and downgrading cython in this environment. However I then ran into problems that pillow was not installed and got myself really confused about what I was doing. I tried installing pillow in the environment but couldn't get it to work. 

What I would try next is to start over fresh and use a legacy version raspberry pi OS, from a time when this was working for others. I assume this comes with Python and all the packages in a compatible way, I would avoid running any updates. 
Not sure when I'm going to try this though.

WLedden, 2023-10-26:
> Unfortunately not. I tried using an environment with Python3.9 (as that seemed to work for others previously) and downgrading cython in this environment. However I then ran into problems that pillow [...]

Will give using an older python venv a go, I've had the pillow issue before, fix I found was using an older version of that too, 9.5.0 version worked for me

ziyadedher, 2023-11-01:
Are you using https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/rgb-matrix.sh to install this library on a RPi? If so, note that the commit referenced in that script (as of https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/a282edbda5de1e9afcbb2bc4cfc5d37710a579e0/rgb-matrix.sh#L10) is really old! It seems to align with your error message of

```rgbmatrix/core.cpp:198:12: fatal error: longintrepr.h: No such file or directory
  198 |   #include "longintrepr.h"```

since in newer versions of this library, that include is behind a check for `PYTHON_MAJOR_VERSION < 3`.

I submitted a PR to the Adafruit repo to bump the commit, but just changing it locally should do the trick for now: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/269

dogerber, 2023-11-14:
I checked this now and @ziyadedher was correct, as this fixed the issue. 
Just make sure to manually delete everything before you run the rgb-matrix.sh again (even tough it says it overwrites, for me it didn't).

This issue can be closed, thanks a lot @ziyadedher!

closed 2025-11-27 by marcmerlin