hzeller/rpi-rgb-led-matrix #1529: 64x64 display: beginner wiring question

Esshahn, 2023-04-06:
I am confused as to the wiring. I have one 64x64 display matrix and a Raspberry Pi3. I've tried:
- the wiring table with the smileys, with strobe=7 (runnind D 0 gives red leds displaying random data but no the example)
- the "regular" column in the mapping table, with strobe=4 (which does not produce any output at all)

I've rewired everything a dozen times and the Pi seems to work fine.
Any help is greatly appreciated.
Thank you for this lib.

davemaster, 2023-04-06:
[image]

Above, the REGULAR, also works with [dragon adapter ](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/)
As you're trying with RPi 3, try to slowdown = 2 the signal speed

Here an example ( try this, with ONLY ONE RGB MODULE), check if you have at least a 5A power supply

sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=64 --led-gpio-mapping = "regular" --led-slowdown-gpio=2 --led-multiplexing=0

You can try with many --led-multiplexing option values 0(default) - 12 (i think, or more) according your type of RGB module

Esshahn, 2023-04-11:
thank you @davemaster for the support. I've rewired everything according to the specs above and used the same commands provided. However, I could only produce the same garbage output of randombly flashing LEDS 

[image]

I'm beginning to think that the display itself might be broken... 
Any thoughts what I could try next? 
Thank you very much :)

Esshahn, 2023-04-11:
With
`sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=5 --led-multiplexing=17` I was able to produce slightly better results. I can now somewhat see the rotating cube, but only barely.
- The output is always and only (in all demos) just red 
- The output seems to be split in 4 sections that are not in logical order

Hope someone has seen this already and can give me a hint.
Cheers.

Esshahn, 2023-04-11:
....and my final update, as it works now :)

With the previous issues solved one by one, I switched the command back to the default and left out the multiplexing setting. With `sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=5` everything's fine now.

[image]

combs, 2023-04-23:
nice! probably makes sense to close this issue...

closed 2023-04-25 by hzeller