hzeller/rpi-rgb-led-matrix #1530: Python API – problem importing rgbmatrix after switching to Code on rapsberry pi

ElBertrando, 2023-04-06:
I have a functioning project for over 2 years, but I recently switch from Code-OSS to Visual Studio Code as I see that is now available for raspberry pi 4. The code has not changed, but with Code there is suddenly a problem importing the rgbmatrix folder.
I use samplebase.py —> [https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/samplebase.py](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/samplebase.py) as my model.
I tried recompiling in the rgbmatrix root folder to no avail. 

[image]

Esshahn, 2023-04-11:
My guess is that VSCode is using a different Python version than the one you installed rgbmatrix with. 
You can see the Python version in the bottom of the VSCode window, like this

[image]

It's usually a good idea to work inside a virtual environment to ensure all dependencies are correctly installed.

From within VSCode, open a terminal window and type
`python -m venv myenv` to create a virtual environment named `myenv`
Then activate it by typing `source myenv/bin/activate` 
You should now be inside the virtual environment and can `pip install` all dependencies

Hope that helps.

Esshahn, 2023-04-12:
I'm no Python expert, but from your description it sounds like you've moved the rgbmatrix lib into the location where pip is installed. I would recommend to have them installed in a virtual environment.

ElBertrando, 2023-04-12:
I've created the virtual environment and re-installed all of the packages, however I still receive the error message regarding the import of the _**rgbmatirx**_  folder in regards to the overall API for the [rpi-rgb-led-matrix](https://github.com/hzeller/rpi-rgb-led-matrix) -- I am still a little unclear where this directory should be installed due to the dependencies to folders levels up (or down) as if pertains to the python api. I use the samples as general guide. 
[image]

combs, 2023-04-23:
my 2 cents as a rando--this doesn't seem related to this library in particular and it would be best to close the issue to de-clutter the issue list. 

Creating a virtual environment, activating the virtual env by choosing it in VSCode, and then installing the package inside the VSCode terminal oughta put everything in its right place. But this is ultimately a VSCode/Python/venv thing and nothing in particular with this library.

closed 2023-04-24 by ElBertrando