hzeller/rpi-rgb-led-matrix #1531: 128x64 single panel issue of flickering

dineshimatrix, 2023-04-09:
Hi 

I've using the unilumin 128x64 rgb led display of size 320mmx160mm which is P2.5 and mentioned as 32 scan in the datasheet. I tried the library and i'm facing the flickering issue in python code. Please help me to resolve the issue. i tried multiple settings value and still can't be able to fix the issue. Anyone please help.

my code:
        graphics.DrawText(offscreen_canvas, font4, 0, 24, textColor_green, datetime.now().strftime('%I:%M:%S'))
        graphics.DrawText(offscreen_canvas, font2, 0, 42, textColor_blue, datetime.now().strftime('%A'))
        offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)
        time.sleep(0.5)
        offscreen_canvas.Clear()

combs, 2023-04-23:
there are some Python performance tips here,
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

dineshimatrix, 2023-04-26:
Hi @combs ,

I've tried the performance options and it seems i can get the flicker free only if i reduce the pwm bits to 2.. but it reduces the color quality. Is there any better way to improve? Please help.

anandrajgupta, 2023-04-26:
use c binding instead of python, you will be amazed with the performance