hzeller/rpi-rgb-led-matrix #1532: weird 96x48 panel

kilograham, 2023-04-09:
I just bought a flexible (bendable) 96x48 HUB75 panel, which turns out to useL

```
6 x SM5368PF - row drivers
12 x SM16208S - column drivers
```

I'm not using this library, but since I've been using it at a reference, i thought I'd shared what I have figured out in case anyone else comes across it:

* The 12 column drivers work as normal, shifting in 96 pixels for two rows (RGB0, and RGB1 which deal with the top 24 or bottom 24 pixels of the panel)
* The SM5368PF are shift style row drivers and select zero or more rows.

The weird thing is how the SM5368PF is set up for shifting:

* **A** is the clock
* **C** is the data

There are two chains of 3 SM536PF handling 24 even numbered rows and 24 odd numbered rows each, only one of which is enabled at a time.

on each positive **A** clock edge, a data bit is shifted in to the current (odd/even) chain of shift registers, then the panel flips  to enable that set of odd/even rows, and switches to shift the next bit into the other (even/odd) set of rows.

To add to the fun; the data bit shifted in is the XOR of the current **C** bit with the previous value shifted in on the last clock

----

I'm a bit confused how you are expected to usefully use this for anything more complicated than just shifting a single row pattern down one row at a time.

CamelCaseName, 2023-04-30:
its absolutely normal for these panels to be driven one row a time, mostly though two rows a time. you just have to do it fast enough for it not to flicker

vcdssss, 2025-05-17:
Use this work fine with ESP 32 Wroom 32
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Arduino.h>
#include <string.h>

#define MATRIX_WIDTH 96
#define MATRIX_HEIGHT 48

// HUB 75 pin def
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 22   //Can be changed
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32   // Can be changed
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

// Matrix Config HUB75
HUB75_I2S_CFG mxconfig(MATRIX_WIDTH, MATRIX_HEIGHT);
MatrixPanel_I2S_DMA matrix;

void setup() {
  Serial.begin(115200);

  // pin setted
  mxconfig.gpio.r1  = R1_PIN;
  mxconfig.gpio.g1  = G1_PIN;
  mxconfig.gpio.b1  = B1_PIN;
  mxconfig.gpio.r2  = R2_PIN;
  mxconfig.gpio.g2  = G2_PIN;
  mxconfig.gpio.b2  = B2_PIN;
  mxconfig.gpio.a   = A_PIN;
  mxconfig.gpio.b   = B_PIN;
  mxconfig.gpio.c   = C_PIN;
  mxconfig.gpio.d   = D_PIN;
  mxconfig.gpio.e   = E_PIN;
  mxconfig.gpio.lat = LAT_PIN;
  mxconfig.gpio.oe  = OE_PIN;
  mxconfig.gpio.clk = CLK_PIN;

  // Matrix ini
  matrix.begin(mxconfig);
  matrix.fillScreen(0);
}

// Test 1: Colour cycle
void testColorCycle() {
  uint16_t colors[] = {
    matrix.color565(255, 0, 0),    // Red
    matrix.color565(0, 255, 0),    // Green
    matrix.color565(0, 0, 255),    // Blue
    matrix.color565(255, 255, 0),  // Yellow
    matrix.color565(0, 255, 255),  // Cyan
    matrix.color565(255, 0, 255),  // Magenta
    matrix.color565(255, 255, 255) // White
  };
  for (int i = 0; i < 7; i++) {
    matrix.fillScreen(colors[i]);
    delay(500);
  }
  matrix.fillScreen(0);
  delay(500);
}

// Test 2: Draw lines
void testLines() {
  matrix.fillScreen(0);
  uint16_t color = matrix.color565(255, 255, 0); // Yellow
  int centerX = MATRIX_WIDTH / 2;
  int centerY = MATRIX_HEIGHT / 2;
  matrix.drawLine(0, 0, centerX, centerY, color);
  matrix.drawLine(MATRIX_WIDTH - 1, 0, centerX, centerY, color);
  matrix.drawLine(0, MATRIX_HEIGHT - 1, centerX, centerY, color);
  matrix.drawLine(MATRIX_WIDTH - 1, MATRIX_HEIGHT - 1, centerX, centerY, color);
  delay(2000);
}

// Test 3: Draw circles
void testCircles() {
  matrix.fillScreen(0);
  uint16_t color = matrix.color565(0, 255, 255); // Cyan
  for (int r = 5; r < MATRIX_HEIGHT/2; r += 3) {
    matrix.drawCircle(MATRIX_WIDTH/2, MATRIX_HEIGHT/2, r, color);
    delay(200);
  }
  delay(2000);
}

// Test 4: Bue ball animation
void testBouncingBall() {
  int ballX = MATRIX_WIDTH / 2;
  int ballY = MATRIX_HEIGHT / 2;
  int ballRadius = 3;
  int dx = 1;
  int dy = 1;
  unsigned long startTime = millis();
  while (millis() - startTime < 5000) { // run 5 seconds
    matrix.fillScreen(0);
    matrix.fillCircle(ballX, ballY, ballRadius, matrix.color565(255, 255, 255));
    delay(50);
    ballX += dx;
    ballY += dy;
    if (ballX - ballRadius <= 0 || ballX + ballRadius >= MATRIX_WIDTH) {
      dx = -dx;
    }
    if (ballY - ballRadius <= 0 || ballY + ballRadius >= MATRIX_HEIGHT) {
      dy = -dy;
    }
  }
  matrix.fillScreen(0);
}

// Test 5: Scrolling text
void testScrollingText() {
  matrix.fillScreen(0);
  const char* msg = "Test Scrolling Text ";
  int textWidth = strlen(msg) * 6;  
  //Scrolling text
  for (int x = MATRIX_WIDTH; x > -textWidth; x--) {
    matrix.fillScreen(0);
    matrix.setCursor(x, MATRIX_HEIGHT/2 - 4);
    matrix.setTextColor(matrix.color565(0, 255, 0)); // Green
    matrix.print(msg);
    delay(100);
  }
  delay(500);
}

void loop() {
  testColorCycle();
  testLines();
  testCircles();
  testBouncingBall();
  testScrollingText();
}

marcmerlin, 2025-05-20:
I do not see a bug or actionable support request.  If you want to chat about panels, you can use discourse mentioned in the README

closed 2025-05-20 by marcmerlin