hzeller/rpi-rgb-led-matrix #1533: PWM still flickering

brthurr, 2023-04-12:
Hello

Running on a RPi 3. Raspian 11 base, fresh install.

Using the Adafruit Bonnet, pins 4 & 18 jumpered per instructions. 2 Adafruit LED matrices (64x32, chained). 5V power adapter.

Using `sudo python3 simple-square.py --led-rows=32 --lod-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2`, I am still seeing flicker. I have used 1-4 for the gpio values.

Is there anything else I can try to eliminate the flicker? Am I just not pushing enough power?

Thanks.

hzeller, 2023-04-12:
Generally, follow https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting
Using C++ and not Python will also improve flicker as there is less memory churn.

brthurr, 2023-04-14:
Messing with the led refresh limit helped some. I have also reflashed with DietPi. We will see if it improved.

closed 2023-04-14 by brthurr