hzeller/rpi-rgb-led-matrix #1535: Undefined symbol: _Py_ZeroStruct

Solaerts, 2023-04-23:
Hey, so I try to code RGB matrix panel of 64x32 with a Raspberry PI 2, I installed all files needed and when I want to lunch the code runtext.py with Thonny I get this error message: 
ImportError: /home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/core.so: undefined symbol: _Py_ZeroStruct
I really don't know what it means and when I try to run a code i made, Thonny say that he could'nt find the rgbmatrix package. I really don't know what I do wrong if someone can help me it would be great.

hzeller, 2023-04-23:
Did you compile and install the Python according to the [docs](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python) ?

Solaerts, 2023-04-23:
Yeah i think so, I was just about to restant the process from the begining.

combs, 2023-05-07:
nice! Probably makes sense to close this issue out then?

closed 2023-05-07 by hzeller