hzeller/rpi-rgb-led-matrix #1538: Replaced faulty panel and new issue now

bierbarrel, 2023-04-25:
I had a yellow block on one of my 4 panels. Troubleshot down to bad panel. Where ever I moved the panel in the array the yellow block stayed on the new panel. I ordered a new panel. It uses the same chipset. On the new panel I get green when Im supposed to have blue. Is there a way to adjust the panel to get it to color match the other panels? Or do I need to get another board?

thx

anandrajgupta, 2023-04-26:
--led-rgb-sequence

ledvinap, 2023-04-26:
@anandrajgupta : That applies to all panels, but only one has different RGB mapping in this case.

@bierbarrel : You can place new panel last and swap RGB wires ...

bierbarrel, 2023-04-26:
are you talking about swapping wires in the ribbon cable?

bierbarrel, 2023-04-26:
[image]

I have the screen separator lines set to red but the bad panel shows blue

ledvinap, 2023-04-26:
@bierbarrel : Some options are:
- separate and swap wires before crimping PFL
- cut RGB wires and solder them in correct order 
- use (a lot of) premade jumper wires (hole-pin to display or pin-pin between 2 cables)

It is possible to do it in software (https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L337), but it is non-trivial to do in in clean way and `Framebuffer::Fill()` won't work correctly.

bierbarrel, 2023-04-26:
Thx petr.. I'll give that a shot. I can make up a ribbon cable. I will just have to figure out the RGB lines.

bierbarrel, 2023-04-29:
Up and running....thx guys!
[image]
[image]

bierbarrel, 2023-04-29:
Ill make a ribbon cable later...

combs, 2023-05-07:
nice! Probably makes sense to close this issue out then?

closed 2023-05-08 by bierbarrel