hzeller/rpi-rgb-led-matrix #1541: Cannot make 48x48 led matrix

wash34000, 2023-05-01:
Hello,

I get some old matrix led 48x48 with MBI5154 chip on it.

I can't get it to work properly. I could see that it was made up of two 24x48 matrix with a connector for each part. I tested the demo and I only have a 16x24 area that displays on the right, the left area of 16x24 has a line of 4 and the bottom area of 8x48 does not display anything.

The weirdest thing is the connector part, it's not a standard 16-pin port but a 20-pin port. and on this connection I have R3 G3 B3 that I don't know how to use.

I also only have AB, so I tried with the setting "--led-row-addr-type=1" but the result is worse than with 0.

The best rendering was obtained with this: "sudo ./demo -D 1 runtext.ppm --led-rows=16 --led-cols=48 --led-row-addr-type=0 --led-multiplexing =13"

Pinout:
  1 - R1             2 - G1
  3 - B1             4 - GND
  5 - R2             6 - G2
  7 - B2             8 - GND
  9 - R3           10 - G3
11 - B3           12 - GND
13 - A             14 - B
15 - NC          16 - NC
17 - CLK         18 - LATCH
19 - OE           20 - GND

Thanks in advance

wash34000, 2023-05-01:
I've try to switch pin from R2 / G2 / B2 to R3 / G3 / B3 and with this wiring i have the bottom area on the rightr working but no more the center. 
So now i'm sure that i need to connect all 3 datas to make it working ...

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828