hzeller/rpi-rgb-led-matrix #1547: "undefined reference" after copy code

Fipschen, 2023-05-19:
Hello together,

two years ago I writed a code for my led matrix in Python. Now, because of performence I'd like zu write it in C++. I'm new and a noob in C++.
I'd try to take existing code and change on after another little thinks until I'm at my project.
But I'm failing at the beginning: I'd copy the code of /utils/text-scroller.cc and when I'll compile it, I get only "unfefined reference to[...]" errors:
`/usr/bin/ld: /tmp/ccxV8ntJ.o: in function `usage(char const*)':
main.cpp:(.text+0xb0): undefined reference to `rgb_matrix::RGBMatrix::Options::Options()'
/usr/bin/ld: main.cpp:(.text+0xbc): undefined reference to `rgb_matrix::RuntimeOptions::RuntimeOptions()'
/usr/bin/ld: main.cpp:(.text+0xd0): undefined reference to `rgb_matrix::PrintMatrixFlags(_IO_FILE*, rgb_matrix::RGBMatrix::Options const&, rgb_matrix::RuntimeOptions const&)'
/usr/bin/ld: /tmp/ccxV8ntJ.o: in function `main':
main.cpp:(.text+0x208): undefined reference to `rgb_matrix::RGBMatrix::Options::Options()'
/usr/bin/ld: main.cpp:(.text+0x214): undefined reference to `rgb_matrix::RuntimeOptions::RuntimeOptions()'
/usr/bin/ld: main.cpp:(.text+0x230): undefined reference to `rgb_matrix::ParseOptionsFromFlags(int*, char***, rgb_matrix::RGBMatrix::Options*, rgb_matrix::RuntimeOptions*, bool)'
/usr/bin/ld: main.cpp:(.text+0x670): undefined reference to `rgb_matrix::Font::Font()'
/usr/bin/ld: main.cpp:(.text+0x680): undefined reference to `rgb_matrix::Font::LoadFont(char const*)'
/usr/bin/ld: main.cpp:(.text+0x6c8): undefined reference to `rgb_matrix::RGBMatrix::CreateFromOptions(rgb_matrix::RGBMatrix::Options const&, rgb_matrix::RuntimeOptions const&)'
/usr/bin/ld: main.cpp:(.text+0x748): undefined reference to `rgb_matrix::RGBMatrix::SetPWMBits(unsigned char)'
/usr/bin/ld: main.cpp:(.text+0x770): undefined reference to `rgb_matrix::RGBMatrix::CreateFrameCanvas()'
/usr/bin/ld: main.cpp:(.text+0x7ac): undefined reference to `rgb_matrix::Font::CharacterWidth(unsigned int) const'
/usr/bin/ld: main.cpp:(.text+0x8b0): undefined reference to `rgb_matrix::DrawText(rgb_matrix::Canvas*, rgb_matrix::Font const&, int, int, rgb_matrix::Color const&, rgb_matrix::Color const*, char const*, int)'
/usr/bin/ld: main.cpp:(.text+0x92c): undefined reference to `rgb_matrix::RGBMatrix::SwapOnVSync(rgb_matrix::FrameCanvas*, unsigned int)'
/usr/bin/ld: main.cpp:(.text+0x974): undefined reference to `rgb_matrix::Font::~Font()'
/usr/bin/ld: main.cpp:(.text+0x994): undefined reference to `rgb_matrix::Font::~Font()'
collect2: error: ld returned 1 exit status`

It's to 100 % the code from utils/text-scroller.cc.
The only diferent is the path to file:
/directory/main.cpp
/rpi-rgb-led-matrix/[...]
But the include must be right, I'll don't get a include-error.

Can someone help me..?

Thanks

Fipschen

hzeller, 2023-05-19:
Looks like your code does not link the library.

Fipschen, 2023-05-19:
Thank you *facepalm*
I tought to set the include is enough. Now it works.

Thank you also for the library :)

closed 2023-05-19 by Fipschen