hzeller/rpi-rgb-led-matrix #1548: Trouble seeing correctly simple-square sample on 96x96 display

ignaciogaticarojas, 2023-05-22:
Hello ,

Im having trouble finding a way to make simple-square sample work well on a 96x96 screen display made out two 96x48 matrix 
 16208 chipset panels using the U-Mapper. Running it with : sudo python3 simple-square.py --led-cols=96 --led-slowdown-gpio=3 --led-gpio-mapping=adafruit-hat  --led-rows=48 --led-chain=2 --led-pixel-mapper=U-mapper  --led-parallel=1 --led-brightness=70

I tried different mappers and all giving me abstract shapes.

Please let me know what would be an option to fix this 

here is an image of the output.

[image]

hzeller, 2023-05-22:
did you also go through all `--led-row-addr-type` ?

ignaciogaticarojas, 2023-05-22:
--led-row-addr-type=3 made it thank you again @hzeller

closed 2023-05-22 by ignaciogaticarojas