hzeller/rpi-rgb-led-matrix #1551: text-scroller speed abruptly changes

cyphunk, 2023-05-29:
I've not yet figured out why this happens but the text-scroller utility will consistently after displaying a few text lines slow down to half the set scroll speed (-s). I confirmed that '--led-show-refresh' does not change.

I believe this probably just a result of running this utility in  a multi-threaded envirionment. But if you have any suggestions on how to improve the reliability I'd be interested.

My setup is adafruit-hat-pwm with 6 panels of 64x32 each. Full execute: 
```text-scroller \
  --led-brightness 100 \
  --led-gpio-mapping=adafruit-hat-pwm \
  --led-rows 32 --led-cols 64 --led-chain=6 \
  --led-no-drop-privs \
  --led-pixel-mapper="S-Mapper;Rotate:90" \
  --led-slowdown-gpio=2 \
  -f ourfonts/SourceSansPro-Regular.otf-48.bdf \
  -s 2 -l 1 -C 255,0,0 \
  -y 5 \
  "$PHRASE" \
 --led-show-refresh
```

hzeller, 2023-05-29:
When you say "slower after a few lines", is it that you run the utility for a while, it shows the text, but then the text slows down after a while ? Or you run it with different texts that seem to run with different speeds ?

The text-scroller internal delay should take care of variation in system speed as calculates the next time for a frame as an absolute time, so it should under normal circumstances not change anything in the speed over time.
So unless drawing the font is consistently taking longer than a frame is supposed to last, or something is messing with the system timekeeping speed it should not result in any  noticeable change. Given the speed is a peaceful `-s 2`, rendering is also not likely to be slower than this.

So I can't reproduce what you're observing. Is there something else going on on the system that might explain things ?

cyphunk, 2023-05-30:
I meant that I have one string of text that is about 120 characters wide and it just runs this string in a loop twice. The slow down apperears about half way through the string on the very first time. It does not happen if I use a regular font but does happen if I use a bold font. There's not much happening on the system but its linux and a raspberry pi so one can't assume stability I guess. 

Thank you for your insights here and the description of how speed is calculated. I've learned to work in the limits but I when I have a chance I'll investigate it in code more thoroughly some other time.

closed 2023-05-30 by cyphunk