hzeller/rpi-rgb-led-matrix #1552: matrix not showing any image

Jamewsss, 2023-06-01:
I have installed the library and followed the install for the adafruit software however when i run any code including the first demo, nothing is displayed on the matrix. the code runs fine with no errors and the matrix display is connected properly to the bonnet and pi. Any idea why this is happening?
[image]

anandrajgupta, 2023-06-02:
--led-gpio-mapping=adafruit-hat

Jamewsss, 2023-06-03:
I tried this but still no output. here is what i enter on the rpi terminal: 

sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/demo -D0 --lew-rows=32 --led-cols=32 llled-no-hardware-pulse --led-gpio-mapping=adafruit-hat

Could it be that the display is faulty? it is a hub75 board so there are no compatibility issues.

davemaster, 2023-06-04:
May be the adafruit hat is defective, I suggest to You try the wiring in the official rpi-rgb-led-matrix github... and use ---led-gpio-mapping=regular (is the default gpio mapping, but for be sure...)

anandrajgupta, 2023-06-04:
> I tried this but still no output. here is what i enter on the rpi terminal:
>
> sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/demo -D0 --lew-rows=32 --led-cols=32 llled-no-hardware-pulse [...]

dot is missing after sudo

isatyamks, 2023-06-29:
If you are using I2C communication to connect the matrix display to the Raspberry Pi, check if there are any address conflicts with other connected devices. Run the command sudo i2cdetect -y 1 in the terminal to see a list of connected I2C devices and their addresses. Ensure that the address of your matrix display matches the one specified in your code.