hzeller/rpi-rgb-led-matrix #1559: sshing via VSCode causes significant CPU usage on RPi4

andytheengineerguy, 2023-06-23:
[image]

After ssh-ing into my RPi 4, the CPU is completely maxxed out for minutes at a time. Screenshot shows the culprit processes. Anyone know why this might be happening? I've used VSCode to SSH into many linux machines - some very old and slow - but haven't seen this behaviour before...

jaygarcia, 2023-06-25:
This has absolutely nothing to do this with this library and everything to do with the service that is installed by VSCode.  Similar stuff happens when you do the same with a JetBrains IDE.

hzeller, 2023-06-25:
`ripgrep` is started by vscode apparently which uses 3 cpu cores alone, then another `node` process almost using a full core. The Raspberry Pi does have issues servicing ethernet if the memory bus is in heavy use. Since VSCode seems to run `node` it probably requires a machines that is about 10x the performance than the pi can offer.

If you want to have some ok performance, edit locally on your main machine, then copy resulting code over (e.g. `rsync` or just `scp`.).

Personally, I am using emacs if I want to work remotely on an embedded device. The `tramp mode` allows to seamlessly edit remote files as well as running make etc. (it uses ssh underneath) without taxing the remote machine too much.

Anyway, no problem with this library or anything that can be fixed here.

closed 2023-06-25 by hzeller

andytheengineerguy, 2023-07-06:
Hi @hzeller and @jaygarcia, you are correct, this has nothing to do with this library 😅 My apologies, I was bringing this up alongside DietPi (both for the first time) and got my wires crossed - I intended to post this on the DietPi forum! hahaha.

Regardless, I appreciate your insight @hzeller. Thank you. I will give emacs a try.