hzeller/rpi-rgb-led-matrix #1565: Set up RTC for RGB LED Matrix Panel Drive Board For Raspberry Pi

shinych, 2023-07-03:
Hello (and sorry if I am asking in the wrong forum)

Is there any documentation or hints on how to set up RTC for that [RGB LED Matrix Panel Drive Board from Electrodragon](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
)?
Just putting the switch into RTC position does not work. Is there any driver or so?

[image]

hzeller, 2023-07-03:
You should probably ask the Electrodragon folks who sold it to you.

shinych, 2023-07-03:
Thanks Henner for the prompt reply. And all your work.

The Electrodragon folks explicitly refer to "Please ask for support via github", refering to this project. 
https://w.electrodragon.com/w/RPI_RMP_Guide

Unfortunately I can find no more information on that.

hansu, 2024-06-15:
There are a lot of tutorials if you search for "DS1307 RTC Raspberry Pi", e.g. https://www.instructables.com/Set-up-Real-Time-Clock-RTC-on-Raspberry-Pi/

The only thing I had to do additionally was to send a valid value to the RTC because the reading for seconds resulted in  0x80 which is not valid.
Just type with installed i2c-tools:
`sudo i2cset -y 0 0x68 0x00`

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828