hzeller/rpi-rgb-led-matrix #1569: Display error in a 64X64 matrix

TotoInamin, 2023-07-12:
I recently purchased an Adafruit 64x64 matrix and did the E/8 soldering. I am running with a Raspberry Pi and the code is 
`sudo ./demo -D 0 --led-rows=64 --led-cols=64 --led-gpio-mapping=adafruit-hat`
And the display is below and I dont know what is wrong. Thanks much for any help!

[image]
[image]
[image]

anandrajgupta, 2023-07-12:
--led-multiplexing
and
--led-row-addr-type

davemaster, 2023-07-12:
Greetings,

FIRST, wire using [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) without that RGB Matrix bonnet, the add to your command the option --led-multiplexing=2, until 14 I think, or more, always that value is increasing for add more support to many new rgb panels...

anandrajgupta, 2023-07-12:
try this option too
--led-row-addr-type

most probably it will solve your issue...