hzeller/rpi-rgb-led-matrix #1570: What is the maximum amount of matrixes in one chain?

DanyloTymets, 2023-07-17:
Hello, I'm working on the project which involves the use of at least 36 matrixes connected in one chain. Currently, I'm using Raspberry Pi 4 model B 4GB and Adafruit RGB Matrix Bonnet to control 5 matrixes, everything seems to work well, but the goal is to use 36+ matrixes, so my question is whether it is possible?
Regarding project needs, I'm considering usage of Raspberry Pi 4 8Gb and Adafruit RGB Matrix HAT.
A little bit more details, I'm using 16x32 RGB LED P8 matrixes, during the usage, not more than 6 matrixes will be working at once. On the other hand, the project implies a rapid change of displayed data and its location, gained from a text file which refreshes with high frequency. That being said leads to another concern I have, whether the refresh frequency won't drop too much on connection of so many matrixes?

anandrajgupta, 2023-07-17:
in code increase the chain length value to desired number, then run the code, you will see how much frequency you are getting, then insert a panel and see if output is acceptable...

andytheengineerguy, 2023-07-19:
HZeller mentions in the readme the following:

> "Maximum resolutions reasonably achievable: A general rule of thumb is that running 16K pixels (128x128 or otherwise) on a single chain, is already pushing limits and you will have to make tradeoffs in visual quality. 32K pixels (like 128x256) is definitely pushing things and you'll get 100Hz or less depending on the performance options you choose. This puts the maximum reasonable resolution around 100K pixels (like 384x256) for 3 chains."

I can generally confirm this comment, as I am currently controlling two 128x64 panels (16.3K pixels) on a single chain, and reasonable refresh rates are only possible with significant degradation in color quality (~5 or 6 PWM bits instead of 11)

Relating this to your situation of using 16x32 panels, this means that you can in theory have 32 panels per chain before reaching your 16k pixel 'pseudo-maximum' 

To check this, show the refresh rate and set the chain variable to 32; the script should show you the refresh rate even if you don't have the physical panels connected

DanyloTymets, 2023-07-19:
@anandrajgupta, @andytheengineerguy Thank you a lot, that the solution I was looking for!

closed 2023-07-19 by DanyloTymets