hzeller/rpi-rgb-led-matrix #1573: Another 80x40 P4 indoor acting weird

lukee7, 2023-07-23:
Hi!

I think I found another P4 80x40 that is somehow acting weird.

After fiddling around with all the ideas in #986 and #49 and #1499 I finally gave up.
Main problem is: I don't know know anything about the panel. Got it bundled with some NovaStar receiver card that (to my understanding) would need a NovaStar processor and/or some Windows-only Software.

I did

```
./text-example --led-gpio-mapping=adafruit-hat --led-cols=80  --led-rows=40 -f ../fonts/6x10.bdf
1 2 3 4 5 6
```

[image]

Adding `a e i o u` in 2nd and 3rd (2nd line appeared mixed into 123456 text; 3rd line appeared below the numbers) line
[image]

Also the demo 0:
`./demo --led-gpio-mapping=adafruit-hat --led-cols=80  --led-rows=40 -D 0
`
[image]

I appreciate any help - really clueless how to solve the puzzle ^^

thanks,
lukas

anandrajgupta, 2023-07-24:
led-multiplexing?

lukee7, 2023-07-24:
> led-multiplexing?

Tried 0-18, but it didn't get better 🤷‍♂️

lukee7, 2023-07-25:
It's supposed to be this module: https://www.qzxrled.com/outdoor-3d-p4-led-module-led-video-wall

But there's no useful information on their website. I asked for a datasheet right now.

lukee7, 2023-07-28:
[image]

There's a lot of those on the board. Reads "DP51250" (211307 in second line).

lukee7, 2023-08-03:
Well...I gave up. The "pixel mover" example did not help in understanding what's going on.
I ordered a 64x32 panel at Amazon and everything just works out-of-the-box

closed 2023-08-03 by lukee7