hzeller/rpi-rgb-led-matrix #1574: Python binding build issue

chrisgilldc, 2023-07-30:
Good morning. I'm having some issues getting the library to install the python bindings on my Pi. I'm using a Pi4 with the Adafruit Hat and the latest RPiOS Lite (11).
The library builds successfully, and builds the bindings. With this install command:

`sudo make install-python PYTHON=$(command -v python)`

I get the following:

```
/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/usr/local/lib/python3.9/dist-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
zip_safe flag not set; analyzing archive contents...
rgbmatrix.__pycache__.core.cpython-39: module references __file__
rgbmatrix.__pycache__.graphics.cpython-39: module references __file__
```

I then wind up with files in /usr/lib/python3.9/site-packages. Notably, this isn't in the system python search path, which makes getting the library to import rather dirty. Like so:
```
pi@testpi:~ $ python3 -c "import rgbmatrix; print(rgbmatrix.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'rgbmatrix'
pi@testpi:~ $ export PYTHONPATH=/usr/lib/python3.9/site-packages/rgbmatrix-0.0.1-py3.9-linux-aarch64.egg
pi@testpi:~ $ python3 -c "import rgbmatrix; print(rgbmatrix.__version__)"
0.0.1
```

I can set the PYTHONPATH to get around this for the time being, but that strikes me as an inelegant hack I don't love. Oddly, when I installed previously (~Sept 2022), also on RPiOS 11, this wasn't an issue. Something must have changed between then and now, either to this library or an underlying library.

crackerjacques, 2023-10-14:
maybe it will be fixed.

`sudo pip install setuptools==58.2.0`

chrisgilldc, 2023-10-16:
I'll aim to test this in the next couple of days, thanks.

BornTKill, 2024-01-28:
Dear,

Same issue on my side. was on raspbian (armv7)  before then switch to dietpi (aarch64). Any find a solution ? @dogerber

BornTKill, 2024-01-28:
better with 

sudo pip install setuptools==58.2.0

dogerber, 2024-01-29:
> Dear,
> 
> Same issue on my side. was on raspbian (armv7) before then switch to dietpi (aarch64). Any find a solution ? @dogerber

Hi,

I was able to dodge this problem using [this](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1526) solution. But that's only because I was using a adafruit product.

chrisgilldc, 2024-01-29:
Good morning, been a while since I dug into this project.
Staying on the same base RaspiOS version, I was able to get a good install using setuptools 58.2.0 and the adafruit script.
Updating to setuptools 69.0.3 makes the problem reoccur. 
So the downgrade, in my case, works, although it's not a permanent fix.