hzeller/rpi-rgb-led-matrix #1578: ICN2037BP 128x64 ABCDE Panel renders with black bars and incorrect demo images

ajhoss, 2023-08-12:
Hi everyone, I’m new to the hardware side of things and I’ve been trying to get a 128x64 led matrix with ICN2037BP set up with an adafruit + HAT. The output is similar to what is described in https://github.com/hzeller/rpi-rgb-led-matrix/issues/521 issue but I don’t understand the solution. It’s also not using an adafruit HAT so I’m curious if how I can translate it to work with the HAT. Any help would be appreciated because I’m going a little nuts with this. The display output is exactly the same as the first picture in that issue. I have soldered the jumper from E to 8 and done the performance jumper from 4 to 18. But there are black bars on the display. Only the top 1/4 and the 3/4 have output, but the cube demo doesn't render a cube, it renders squares. And 2/4 and 4/4 stay black no matter what settings I've put in. Any help would be appreciated. Thanks!

gguy825, 2023-08-14:
Can you post a picture of the HAT you're using? i recently had the same issue.

gguy825, 2023-08-14:
> Thanks for replying! Sure. Here it is. I don’t have any mods connected currently because I was trying to get back to square one. 

ok we have the same exact one. You need a wire connecting 4 and 18 near the top portion. and also it looks like you soldered the 16 pad near the main connector. I marked it on the edited picture. You need to bridge the solder on E and 8, 16 doesn't need solder so try getting a solder sucker and cleaning that off and bridging the 2 next to it. 

[image]

ajhoss, 2023-08-14:
I did have solder on those but have cleaned them up since to reset. I also had that jumper wire connected previously. Would you mind sharing the arguments you use so I can compare? I’ll get those connections resoldered and then try it again.

gguy825, 2023-08-14:
> I did have solder on those but have cleaned them up since to reset. I also had that jumper wire connected previously. Would you mind sharing the arguments you use so I can compare? I’ll get those connections resoldered and then try it again.

`sudo ./led-image-viewer -f -w3 *.png --led-cols=64 --led-rows=64 --led-chain=2 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 --led-scan-mode=1 --led-pixel-mapper=U -C ` this and some of the demo files like the rotating cube and the clock was what I was using in the beginning without doing the hardware mods. I got flickers and a black bar. The modding fixed these issues and not everything works properly for me. The flickering in the video is from my phones camera, before the modding it the flickering was a lot more noticeable. 

https://github.com/hzeller/rpi-rgb-led-matrix/assets/41169055/7589633d-8da9-42c1-8f2f-9a54c6a0de14

ajhoss, 2023-08-14:
Great! I'll give that a shot tomorrow and report back if things are working. Thanks!

ostee888, 2024-04-03:
Is there any progress on this?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828