hzeller/rpi-rgb-led-matrix #1580: 64x64 ABCDE - black horizontal bar problem

olablt, 2023-08-17:
Testing chain with 3 LED panels:

`sudo ./clock -f ../fonts/10x20.bdf -d "%H:%M:%S" --led-brightness 50 --led-cols 64 --led-rows 64 --led-chain 3 -y 25 -x 50 --led-slowdown-gpio=4`

Result:

[image]

I am using passive adapter https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/passive-3 and modified as per insturctions:

[image]

Back of my panels:

[image]

What could be the problem?

Feuerflamme, 2023-09-02:
i have the same problem

olablt, 2023-09-03:
soldered active board, problem gone

closed 2023-09-03 by olablt