hzeller/rpi-rgb-led-matrix #1583: Using keyboard input for python and raspberry pi 4

pbullion, 2023-08-21:
I'm trying to use the console to change the running variable by just simply hitting enter or something on the keyboard. i keep striking out importing the keyboard from python.

Any ideas? have people run into this before? i get the must be running as root error, even tho i am.
Thanks!
[image]

closed 2023-09-11 by pbullion

Scattercatt, 2024-01-21:
I have run into this issue as well. I'm writing my program in python, and it seems like instantiating the `RGBMatrix` object does something to screw up privileges. Whenever I comment out the line, the `keyboard.read_key()` runs fine with no error. If I have it not commented out, I get the error OP is getting.

hzeller, 2024-01-21:
The privileges are dropped by default to not run as root afterwards. https://github.com/hzeller/rpi-rgb-led-matrix/blob/a3eea997a9254b83ab2de97ae80d83588f696387/include/led-matrix.h#L422-L426
So you can change that by setting a different runtime option (there is an analog in the Python API).

Setting back the privileges should not really be a problem as you only need to be root in the beginning when you set up everything.

Scattercatt, 2024-01-21:
Ah perfect thank you!